openbmc/openbmc #1676: Decrement reboot count on all boot attempts

geissonator, 2017-05-25:
There's currently a bug going on between hostboot and opal where the reboot count is not getting reset when a boot succeeds.  Once that is fixed, we need to change the obmc code to decrement the boot count on all boots.  

Current code only decrements if it hits a host watchdog or checkstop error.  To conform with OP design, the code needs to always decrement the boot count on every boot attempt.  Hostboot will increase the boot count by 1 when doing controlled reboots (i.e. reconfig loops or sbe updates).

geissonator, 2017-07-27:
- [ ] Code in decrement of reboot count on all boot attempts
- [ ] Verify host is now incrementing as expected
- [ ] Ensure automated test team has adequate coverage

# **STATUS**
7/27 - No progress yet

spinler, 2017-07-27:
We also want a few retries when PGOOD fails to to turn on.  Does the boot count process work there too?

geissonator, 2017-07-29:
The design is to give a total of 3 boot attempts for everything (pgood, sbe, host) with the last attempt being done on the other SBE image side.  So yes, this boot count/reattempt will apply to pgood as well.

rfrandse, 2017-08-08:
<https://gerrit.openbmc-project.xyz/5851> Decrement reboot counter on all boot attempts
Resolves  #1676 Decrement reboot count on all boot attempts

closed 2017-08-17 by williamspatrick