openbmc/openbmc #1677: Boot count not getting correctly read and updated by hostboot

geissonator, 2017-05-25:
I've been doing some SBE seeprom redundant side testing and I found a bug of some sort in the path where hostboot does an SBE update and requests a reboot.

Their design is to read the BootCount sensor, increment it by 1, and then set that sensor with the new value.  For some reason, they always get "0" back for this sensor when reading from obmc, even when I know for sure it's at 2 or 3.  

I added some trace on our side to see when it gets set.  This is all we get when they read and set it
```
Apr 20 20:13:08 witherspoon systemd[1]: Started Start host0 SBE.
Apr 20 20:13:08 witherspoon openpower-proc-control[6572]: rebootObj = :1.156
Apr 20 20:13:08 witherspoon openpower-proc-control[6572]: rebootCount=2
…
Apr 20 20:16:15 witherspoon ipmid[861]: Bus: :1.156, Path: /org/openbmc/sensors/host/BootCount, Interface: org.openbmc.SensorValue
Apr 20 20:16:15 witherspoon sensor_manager2.py[2237]: Setting boot count to 1
```

Here's the relevant hostboot trace:
```
84.37492|SBE|I>updateProcessorSbeSeeproms(): Restart Needed (1). Calling preReIplCheck()
84.40026|SBE|>>sbeDoReboot
84.50229|IPMI|rp: queuing sync 10:2d
84.50348|IPMI|dd: I>write ok 10:2d seq 9e len 1
84.69492|IPMI|dd: I>read b2h ok 14:2d seq 9e len 4 cc 0
84.69516|SBE|I>sbeDoReboot: Writing Reboot Sensor Count=1
84.69577|IPMI|rp: queuing async 10:30
84.69578|sbe|System Rebooting To Perform SBE Update
```

Since hostboot sets the correct sensor, it makes me think they must read it correctly which means there's a bug on the obmc side with providing them this sensor value.

tomjoseph83, 2017-08-05:
@geissonator Would bootcount remain in the /org/openbmc path or would it be moved to xyz/openbmc_project path?

geissonator, 2017-08-05:
Bootcount has been moved into new settingsd in xyz namespace.  Talk with Dhruv if you need details, but either way, seems like we have a big in mapping to either boot count.

geissonator, 2017-08-07:
New boot count:
```
[witherspoon]# busctl get-property xyz.openbmc_project.State.Host /xyz/openbmc_project/state/host0 xyz.openbmc_project.Control.Boot.RebootAttempts AttemptsLeft
u 3
```

tomjoseph83, 2017-08-08:
@geissonator Probably the design is different from what you have explained at the beginning of the thread.  I assume the hostboot would decrement the count by 1 instead of incrementing it.

geissonator, 2017-08-09:
Hostboot increments the reboot count, because they don't want their reboot request to count against them (because of the sbe update).  They've indicated though they may just simplify it and just set the boot count to 2 every time they do this instead of trying this read/modify/write operation.

tomjoseph83, 2017-08-11:
@geissonator The missing piece was the Get Sensor reading support for the boot count(RebootAttempts) sensor. If the hostboot has simplified the operation by always do a setting the value to 2 instead of getting, then this issue should be resolved.

tomjoseph83, 2017-09-01:
# **STATUS**

09/01
This should get closed when 2220 is completed.

geissonator, 2017-09-01:
Tracked under #2220

closed 2017-09-01 by geissonator