openbmc/openbmc #1678: Settings/Network interface redesign

dkodihal, 2017-05-26:
Based on a recent discussion about Settings interfaces related to Network - it seemed these interfaces should be covered by the Network interface model. The current Network interfaces likely cover all that we need to capture in a "host network settings" object, but some of the interfaces may need to be split up, plus we may need new ones.

Actions:
- https://gerrit.openbmc-project.xyz/#/c/4205/6/xyz/openbmc_project/Settings/Network/Config.interface.yaml : cover under Network interfaces
- https://gerrit.openbmc-project.xyz/#/c/4205/6/xyz/openbmc_project/Settings/Time/NTP.interface.yaml : cover under network
- The existing network implementation needs to take a change in the object path naming to indicate objects that denote the bmc network config.

geissonator, 2017-06-15:
Ratan actually working this one

rfrandse, 2017-07-16:
<https://gerrit.openbmc-project.xyz/5179> Make changes as per the new settings infrastructure
Resolves  #1678 Settings/Network interface redesign

closed 2017-07-19 by williamspatrick