openbmc/openbmc #1679: Put WDT1 into external_gpio mode

spinler, 2017-05-26:
On Witherspoon we need WDT1 to be put into the 'set external GPIO on watchdog timeout' mode instead of the 'reset system' mode that it defaults to.

Issue #1636 is being used to add a new attribute to sysfs, so when that is done then we'll need code to actually set that attribute.

One idea is to make a udev rule (that only gets installed on systems that have the WDT1 wired to the fan wachdog circuit) that calls a service to set that attribute as soon as the driver is loaded.

bjwyman, 2017-06-06:
There is apparently still some ongoing discussion about Issue #1636. Depending on how things get implemented for that, this issue/change may or may not be needed.

rfrandse, 2017-06-15:
Not doing this here,

closed 2017-06-15 by rfrandse