openbmc/openbmc #1682: Not able to power on the chassis and host on zaius3 system

PavamanSubramaniyam, 2017-05-29:
I am trying to power on the chassis and host on zaius3 hardware.

root@zaius:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

root@zaius:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
root@zaius:~# obmcutil poweron
root@zaius:~# obmcutil power
pgood = 0
state = 0
pgood_timeout = 10
root@zaius:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off

The machine is installed with the Open BMC and PNOR Firmware images as follows:

root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-12"
VERSION_ID="v1.99.6-12-gcd24073"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-12"
BUILD_ID="v1.99.6"

root@zaius:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%open-power-zaius-v1.17-11-ge962522
 buildroot-2017.02.2-7-g23118ce
 skiboot-5.6.0
 hostboot-5e1b4a6
 linux-4.11.2-openpower1-pef1c0df
 petitboot-v1.4.2-p2462045
 machine-xml-bc19a36
 occ-bc34ad5
 hostboot-binaries-92195a3
 capp-ucode-9c73e9f
 sbe-d770027
[==================================================] 100%

PavamanSubramaniyam, 2017-05-29:
[dmesg.txt](https://github.com/openbmc/openbmc/files/1035642/dmesg.txt)

[journalctl.txt](https://github.com/openbmc/openbmc/files/1035659/journalctl.txt)
[systemctl.txt](https://github.com/openbmc/openbmc/files/1035672/systemctl.txt)

PavamanSubramaniyam, 2017-05-29:
Machine access details are as follows:
=========================
ssh root@zaius3.aus.stglabs.ibm.com (0penBmc)

williamspatrick, 2017-05-29:
```
Apr 20 19:15:33 zaius kernel:  fsi0: can't read slave 00:00 -5
Apr 20 19:15:33 zaius kernel[809]:  fsi0: can't read slave 00:00 -5
Apr 20 19:15:33 zaius systemd[1]: Started Phosphor Fan Monitor Daemon.
Apr 20 19:15:33 zaius systemd[1]: Started Phosphor Fan Control Daemon.
Apr 20 19:15:33 zaius systemd[1]: Reached target Fan Control Ready.
Apr 20 19:15:33 zaius systemd[1]: Started Scan FSI devices.
Apr 20 19:15:34 zaius systemd[1]: Starting Run VCS workaround on host0...
Apr 20 19:15:34 zaius openpower-proc-control[1210]: Failed to open the device.
Apr 20 19:15:34 zaius systemd[1]: vcs_workaround@0.service: Main process exited, code=exited, status=255/n/a
Apr 20 19:15:34 zaius systemd[1]: Failed to start Run VCS workaround on host0.
```

Looks like an FSI issue.

amboar, 2017-05-30:
@PavamanSubramaniyam 

Please use the triple  back-tick markdown when pasting code or console output into github issues. It prevents github's rendering engine interpreting your log output as markdown. What you write looks like:

    ```
    My code, console or log output here
    ```

mine260309, 2017-06-06:
Meet the same issue on Romulus with dev-4.10 kernel, fsi-scan does not setup FSI link and `openpower-proc-control` fails to run VCS workaround.

gtmills, 2018-05-23:
@mine260309 @PavamanSubramaniyam Is this still a problem? Can we close?

closed 2019-04-27 by stale[bot]