openbmc/openbmc #1684: Persist and restore OpenBMC error logs

dkodihal, 2017-05-29:
Persist and restore OpenBMC error logs - Use Cereal for serialization and unserialization.

dkodihal, 2017-06-01:
- [ ] Write recipe to install cereal API headers from https://github.com/USCiLab/cereal to /usr/include/cereal. Have phosphor-logging's recipe depend on this new cereal recipe.
- [ ] Implement cereal load and save methods in the logging Entry class to respectively handle deserialization and serialization.
- [ ] Write serialized error objects to an output file in a persistent location. The serialization can happen right after an error d-bus object is created.
- [ ] When the logging process starts, deserialize error objects, if any, and re-construct Entry objects.
- [ ] Write a unit test to ensure the serialized object matches the Entry object.

rfrandse, 2017-06-14:
<https://gerrit.openbmc-project.xyz/4783> serialization: add tests
Resolves  #1684 Persist and restore OpenBMC error logs

closed 2017-06-26 by williamspatrick