openbmc/openbmc #1686: [Barrelye/Palmetto] pflash Couldn't initialise architecture flash structures

swe12345, 2017-05-30:
Couldn't initialise architecture flash structures
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-12"
VERSION_ID="v1.99.6-12-gcd24073"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-12"
BUILD_ID="v1.99.6"
```
```
root@barreleye:~# uname -a
Linux barreleye 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Fri May 26 23:23:33 UTC 2017 armv5tejl GNU/Linux
```
```
root@barreleye:~# pflash -r /tmp/out.txt -P VERSION
Couldn't initialise architecture flash structures
```
```
root@barreleye:~# pflash -i
Couldn't initialise architecture flash structures
```
```
root@barreleye:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     247676         0    247676   0% /dev
tmpfs                   248964     14276    234688   6% /run
/dev/mtdblock4           13696     13696         0 100% /run/initramfs/ro
/dev/mtdblock5            4096       320      3776   8% /run/initramfs/rw
cow                       4096       320      3776   8% /
tmpfs                   248964         0    248964   0% /dev/shm
tmpfs                   248964         0    248964   0% /sys/fs/cgroup
tmpfs                   248964         0    248964   0% /tmp
tmpfs                   248964        88    248876   0% /var/volatile
```
```
root@barreleye:~# mount
dev on /dev type devtmpfs (rw,relatime,size=247676k,nr_inodes=61919,mode=755)
sys on /sys type sysfs (rw,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
/dev/mtdblock4 on /run/initramfs/ro type squashfs (ro,relatime)
/dev/mtdblock5 on /run/initramfs/rw type jffs2 (rw,relatime)
cow on / type overlay (rw,relatime,lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev)
tmpfs on /var/volatile type tmpfs (rw,relatime)
```

swe12345, 2017-05-30:
Palmetto:
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-12"
VERSION_ID="v1.99.6-12-gcd24073"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-12"
BUILD_ID="v1.99.6"
```
```
root@palmetto:~# uname -a
Linux palmetto 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Fri May 26 23:23:54 UTC 2017 armv5tejl GNU/Linux
```
```
root@palmetto:~# pflash -r /tmp/out.txt -P VERSION
Couldn't initialise architecture flash structures
```
```
root@palmetto:~# pflash -i
Couldn't initialise architecture flash structures
```
```
root@palmetto:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
dev                     247672         0    247672   0% /dev
tmpfs                   248960     38192    210768  15% /run
/dev/loop0               13568     13568         0 100% /run/initramfs/ro
cow                     248960     38192    210768  15% /
tmpfs                   248960         0    248960   0% /dev/shm
tmpfs                   248960         0    248960   0% /sys/fs/cgroup
tmpfs                   248960         0    248960   0% /tmp
tmpfs                   248960        60    248900   0% /var/volatile
```
```
root@palmetto:~# mount
dev on /dev type devtmpfs (rw,relatime,size=247672k,nr_inodes=61918,mode=755)
sys on /sys type sysfs (rw,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
/dev/loop0 on /run/initramfs/ro type squashfs (ro,relatime)
cow on / type overlay (rw,relatime,lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run/initramfs/rw/work)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
tmpfs on /var/volatile type tmpfs (rw,relatime)
```

swe12345, 2017-05-30:
Facing the same issue with the driver v1.99.6-18-g221ab78
```
root@palmetto:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-18"
VERSION_ID="v1.99.6-18-g221ab78"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-18"
BUILD_ID="v1.99.6"
```
```
root@palmetto:~# pflash -r /tmp/out.txt -P VERSION
Couldn't initialise architecture flash structures
root@palmetto:~#
````

swe12345, 2017-05-30:
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-18"
VERSION_ID="v1.99.6-18-g221ab78"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-18"
BUILD_ID="v1.99.6"
```
```
root@barreleye:~# pflash -r /tmp/out.txt -P VERSION
Couldn't initialise architecture flash structures
```

gkeishin, 2017-05-30:
@anoo1 @geissonator  We looked at and this looks like a P8 related issue.  On Witherspoon we don't see this one..

anoo1, 2017-05-30:
On P8 systems, the pnor mtd is missing:
```
root@barreleye:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 02000000 00010000 "bmc"
mtd1: 00060000 00010000 "u-boot"
mtd2: 00020000 00010000 "u-boot-env"
mtd3: 00440000 00010000 "kernel"
mtd4: 01740000 00010000 "rofs"
mtd5: 00400000 00010000 "rwfs"
root@barreleye:~# 
```

Attaching the barreleye journal as well, an excerpt:
[barreleye_journal.txt](https://github.com/openbmc/openbmc/files/1038610/barreleye_journal.txt)

```
Apr 21 19:21:10 barreleye kernel[723]: 5 ofpart partitions found on MTD device bmc
Apr 21 19:21:10 barreleye kernel[723]: Creating 5 MTD partitions on "bmc":
Apr 21 19:21:10 barreleye kernel[723]: 0x000000000000-0x000000060000 : "u-boot"
Apr 21 19:21:10 barreleye kernel[723]: 0x000000060000-0x000000080000 : "u-boot-env"
Apr 21 19:21:10 barreleye kernel[723]: 0x000000080000-0x0000004c0000 : "kernel"
Apr 21 19:21:10 barreleye kernel[723]: 0x0000004c0000-0x000001c00000 : "rofs"
Apr 21 19:21:10 barreleye kernel[723]: 0x000001c00000-0x000002000000 : "rwfs"
```

eddiejames, 2017-06-01:
This is fixed in the newest 4.10. We need to update openbmc to latest 4.10 commits from Joel.

# uname -a
Linux barreleye 4.10.5-fc27159ec6f4c193d04f1430378ccc25a10bd638 #3 Thu Jun 1 11:39:49 CDT 2017 armv5tejl GNU/Linux
# pflash -i
Flash info:
-----------
Name          = /dev/mtd6
Total size    = 64MB 
Erase granule = 4KB 

TOC@0x00000000 Partitions:
-----------
ID=00            part 00000000..00001000 (actual=00001000) 
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           GUARD 0002c000..00031000 (actual=00005000) [ECC]
ID=03             HBD 00031000..00091000 (actual=00060000) [ECC]
ID=04           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=05            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=06            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=07             HBI 001b1000..00751000 (actual=005a0000) [ECC]
ID=08            SBEC 00751000..007e1000 (actual=00090000) [ECC]
ID=09             SBE 007e1000..00829000 (actual=00048000) [ECC]
ID=10            WINK 00829000..00949000 (actual=00120000) [ECC]
ID=11            HBRT 00949000..00ca9000 (actual=00360000) [ECC]
ID=12         PAYLOAD 00ca9000..00da9000 (actual=00100000) 
ID=13      BOOTKERNEL 00da9000..01ca9000 (actual=00f00000) 
ID=14        ATTR_TMP 01ca9000..01cb1000 (actual=00008000) 
ID=15       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=16             OCC 01cb9000..01dd9000 (actual=00120000) [ECC]
ID=17            TEST 01dd9000..01de2000 (actual=00009000) [ECC]
ID=18           NVRAM 01de2000..01e72000 (actual=00090000) 
ID=19         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=20         BMC_INV 01e75000..01e7e000 (actual=00009000) 
ID=21            CAPP 01e7e000..01ea2000 (actual=00024000) [ECC]
ID=22         SECBOOT 01ea2000..01ec6000 (actual=00024000) [ECC]
ID=23     IMA_CATALOG 01ec6000..01ecf000 (actual=00009000) [ECC]
ID=24             HBB 01f60000..01ff0000 (actual=00090000) [ECC]
ID=25         VERSION 01ff7000..01ff8000 (actual=00001000) 
ID=26     BACKUP_PART 01ff8000..02000000 (actual=00000000) 
ID=27      OTHER_SIDE 02000000..02008000 (actual=00000000) 

TOC@0x02000000 Partitions:
-----------
ID=00            part 02000000..02001000 (actual=00001000) 
ID=01            HBEL 00008000..0002c000 (actual=00024000) [ECC]
ID=02           DJVPD 00091000..000d9000 (actual=00048000) [ECC]
ID=03            MVPD 000d9000..00169000 (actual=00090000) [ECC]
ID=04            CVPD 00169000..001b1000 (actual=00048000) [ECC]
ID=05        ATTR_TMP 01ca9000..01cb1000 (actual=00008000) 
ID=06       ATTR_PERM 01cb1000..01cb9000 (actual=00008000) [ECC]
ID=07           NVRAM 01de2000..01e72000 (actual=00090000) 
ID=08         FIRDATA 01e72000..01e75000 (actual=00003000) [ECC]
ID=09         BMC_INV 01e75000..01e7e000 (actual=00009000) 
ID=10         SECBOOT 01ea2000..01ec6000 (actual=00024000) [ECC]
ID=11             HBD 02008000..02068000 (actual=00060000) [ECC]
ID=12             HBI 02188000..02728000 (actual=005a0000) [ECC]
ID=13            SBEC 02728000..027b8000 (actual=00090000) [ECC]
ID=14             SBE 027b8000..02800000 (actual=00048000) [ECC]
ID=15            WINK 02800000..02920000 (actual=00120000) [ECC]
ID=16            HBRT 02920000..02c80000 (actual=00360000) [ECC]
ID=17         PAYLOAD 02c80000..02d80000 (actual=00100000) 
ID=18      BOOTKERNEL 02d80000..03c80000 (actual=00f00000) 
ID=19             OCC 03c80000..03da0000 (actual=00120000) [ECC]
ID=20            CAPP 03da0000..03dc4000 (actual=00024000) [ECC]
ID=21     IMA_CATALOG 03dc4000..03dcd000 (actual=00009000) [ECC]
ID=22             HBB 03f67000..03ff7000 (actual=00090000) [ECC]
ID=23         VERSION 03ff7000..03ff8000 (actual=00001000) 
ID=24     BACKUP_PART 03ff8000..04000000 (actual=00000000) 
ID=25      OTHER_SIDE 00000000..00008000 (actual=00000000)

gkeishin, 2017-06-01:
@swe12345 Can you update to latest build and verify this.. Thanks

geissonator, 2017-06-01:
If we need Joel's version of 4.10, that is not in a build yet.

swe12345, 2017-06-01:
Facing the same issue with Driver -v1.99.6-28-gf6442e1

```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-28"
VERSION_ID="v1.99.6-28-gf6442e1"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-28"
BUILD_ID="v1.99.6"
```
```
root@barreleye:~# pflash -i
Couldn't initialise architecture flash structures
```

gkeishin, 2017-06-01:
@swe12345 Yeah we may need to wait for the new kernel build changes..

rfrandse, 2017-06-03:
<https://gerrit.openbmc-project.xyz/4565> linux: bump version
Resolves  #1686 [Barrelye/Palmetto] pflash Couldn't initialise architecture flash structures

swe12345, 2017-06-13:
Observing the same issue with the Driver v1.99.6-104-gb9feb50
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-104"
VERSION_ID="v1.99.6-104-gb9feb50"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-104"
BUILD_ID="v1.99.6"
```
```
root@barreleye:~# pflash -i
Couldn't initialise architecture flash structures
```
```
root@barreleye:~# pflash -r /tmp/out.txt -P VERSION
Couldn't initialise architecture flash structures
```

gkeishin, 2017-06-13:
This is looking bad for already few builds ... anything on this @eddiejames

swe12345, 2017-06-13:
Stands same with the latest driver v1.99.6-118-g786908a
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-118"
VERSION_ID="v1.99.6-118-g786908a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-118"
BUILD_ID="v1.99.6"
```
```
root@barreleye:~# obmcutil state
CurrentBMCState:    xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState:  xyz.openbmc_project.State.Chassis.PowerState.Off
CurrentHostState:   xyz.openbmc_project.State.Host.HostState.Off
```
```
root@barreleye:~# pflash -i
Couldn't initialise architecture flash structures
```
```
root@barreleye:~# pflash -r /tmp/out.txt -P VERSION
Couldn't initialise architecture flash structures
```

swe12345, 2017-06-14:
With Driver v1.99.6-130-g35cfa84
Driver info
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-130"
VERSION_ID="v1.99.6-130-g35cfa84"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-130"
BUILD_ID="v1.99.6"
```
```
root@barreleye:~# pflash -r /tmp/out.txt -P VERSION
Couldn't initialise architecture flash structures
```

swe12345, 2017-06-15:
Driver info
```
root@barreleye:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-141"
VERSION_ID="v1.99.6-141-ge662190"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-141"
BUILD_ID="v1.99.6"
```
```
root@barreleye:~# pflash -r /tmp/out.txt -P VERSION
Couldn't initialise architecture flash structures
```
```
root@barreleye:~# pflash -i
Couldn't initialise architecture flash structures
```

closed 2017-06-16 by williamspatrick