openbmc/openbmc #1687: Pulse the fan watchdog reset line

spinler, 2017-05-30:
After the Witherspoon fan watchdog has tripped due to a pulse from the WDT1 expiration signal, we'll need to pulse the CLEAR_PULSE GPIO which clocks a flip flop to allow the full-speed signal to the fan controller to be deasserted so the fans can go back to normal.

We're going to run this on a power on before fan control starts.  We never try to recover a tripped watchdog at runtime due to the complexity of all the apps that are involved.

spinler, 2017-05-30:
- [ ]  Create a phosphor-gpio-util tool in the phosphor-gpio-monitor repo that can pulse a GPIO.  It will take command line args to say which GPIO and what to do.
- [ ] Create service file to run it during obmc-chassis-poweron before fan control

rfrandse, 2017-06-06:
<https://gerrit.openbmc-project.xyz/4619> Install phosphor-reset-fan-watchdog service
Resolves  #1687 Pulse the fan watchdog reset line

closed 2017-06-14 by williamspatrick