openbmc/openbmc #1688: Create OnFailure unit monitor

spinler, 2017-05-30:
The fan watchdog needs to be stopped when the fan related services fail enough that they stay in the failed state.

spinler, 2017-05-30:
- [ ] Create a phosphor-unit-failure program that gets the ActiveState property for a unit and then stops or starts another unit if that property = 'failed'.  Command line args would pass in the units and action.
- [ ] Create a phosphor-fan-unit-failure@.service that will start the 'stop-fan-watchdog' target and pass the unit to check with a %i.
- [ ] On the fan related services, add OnFailure=phosphor-fan-unit-failure@%n.service so the program gets invoked with the failing unit name
- [ ] Use the systemd override files to only apply this to the max31785 hwmon instance, not all hwmons
- [ ] Also update the StartLimitIntervalSec and StartLimitBurst fields in these services to configure how many times they can fail before the watchdog would get stopped.

rfrandse, 2017-06-14:
<https://gerrit.openbmc-project.xyz/4740> Monitor fan apps for watchdog protection
Resolves  #1688 Create OnFailure unit monitor

closed 2017-07-07 by williamspatrick