openbmc/openbmc #1689: [Witherspoon]:REST Logging attribute "Resolved" can't be set from external

cdeadmin, 2017-05-30:


gkeishin, 2017-05-30:
Original defect https://github.com/openbmc/openbmc/issues/1640

ojayanth, 2017-08-10:
It is working good in the latest build , also noticed the curl command was wrong.

jayanths-MacBook-Pro:~ ojayanth$ curl -c cjar -b cjar -k -H "Content-Type: application/json" -d "{\"data\":1}" -X PUT https:/9.3.185.33/xyz/openbmc_project/logging/entry/43/attr/Resolved

{
 "data": null,
 "message": "200 OK",
 "status": "ok"

}jayanths-MacBook-Pro:~ ojayanth$ curl -c cjar -b cjar -k https://9.3.185.33/xyz/openbmc_project/logging/entry/43
{
 "data": {
    "AdditionalData": [],
    "Id": 43,
    "Message": "xyz.openbmc_project.Common.Error.InternalFailure",
    "Resolved": 1,   -----------> setting to 1
    "Severity": "xyz.openbmc_project.Logging.Entry.Level.Error",
    "Timestamp": 1501775038378,
    "associations": []
 },
 "message": "200 OK",
 "status": "ok"

gkeishin, 2017-08-10:
working good on latest master.. will close up CQ once in GHE

closed 2017-08-10 by gkeishin