openbmc/openbmc #1690: [Barreleye latest] /tmp/esel messages lost

yadron, 2017-05-30:
If hardware problem happened during booting host, then ESEL messages being received by OpenBMC (using ipmid). The file /tmp/esel created as result. The problem that this ESEL event doesn't exist in BMC:
1. If I restart BMC, it is lost.
2. How to retrieve it? The "curl -b cjar -k https://172.17.27.146/org/openbmc/records/events/" or "busctl tree xyz.openbmc_project.Logging | cat" don't see ESEL events, so currently all ESEL events being lost at BMC.

yadron, 2017-06-20:
@williamspatrick ,
why
`/org/openbmc/records/events/*
` not available anymore?

It looks like it was added on Feb 2016:
https://lists.ozlabs.org/pipermail/openbmc/2016-February/001729.html
Why already deleted if added just some months ago?

I see new xyz that is

```
busctl introspect xyz.openbmc_project.Logging /xyz/openbmc_project/logging
curl -b cjar -k https://172.17.27.28/xyz/openbmc_project/logging/enumerate
curl -b cjar -k https://172.17.27.28/xyz/openbmc_project/logging/internal/manager
```
, but looks like it doesn't have similar functions that were before:

```
List all events
curl -k https://bmc/org/openbmc/records/events/

Create an event log for testing purposes
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": []}" https://bmc/org/openbmc/records/events/action/acceptTestMessage

Delete an event
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": []}" https://bmc/org/openbmc/records/events/1/action/delete

Delete all events
curl -c cjar -b cjar -k -H "Content-Type: application/json" -X POST -d "{\"data\": []}" https://bmc/org/openbmc/records/events/action/clear
```

The only source of info regarding xyz that I've found:

https://github.com/openbmc/openbmc/issues/1297

williamspatrick, 2017-06-28:
Persisting all BMC logs was added recently with #1684 .  They have not been in the /org namespace for quite a while now.  They are under `/xyz/openbmc_project/logging/entry` with the interface documentation being at https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Logging/Entry.interface.yaml

gtmills, 2018-08-20:
@yadron @AlexanderAmelkin Can this be closed?

AlexanderAmelkin, 2018-08-21:
@gtmills, I believe so.

gtmills, 2018-08-21:
Closing. Please reopen if still an issue. Thanks!

closed 2018-08-21 by gtmills