openbmc/openbmc #1691: Need to know  the state of uploading an image

causten, 2017-05-30:
The xyz.openbmc_project.Software.Download service does not have an indication of state.  How do I know when the DownloadViaTFTP has completed (good or bad path)?  As a system admin I would need to know if I sent the correct ip and file name.

amboar, 2017-08-31:
@rfrandse this isn't a question, but a bug. For instance:

```
$ curl -b cjar -k -H "Content-Type: application/json" -X POST           -d '{"data": ["arj/witherspoon-sequoia.pnor.squashfs.tar", "<IP>"]}'           https://<bmc>/xyz/openbmc_project/software/action/DownloadViaTFTP
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
}2 15:48:08 andrew@keelia:~$
```
But the journal reports:

```
Sep 02 15:13:26 witherspoon phosphor-download-manager[940]: Downloading via TFTP
Sep 02 15:13:26 witherspoon phosphor-gevent[1349]: ::8099:c276:8e91:f402%1110813460 - - [2017-09-02 15:13:26] "POST /xyz/openbmc_project/software/action/DownloadViaTFTP HTTP/1.1" 200 169 0.449432
Sep 02 15:13:26 witherspoon phosphor-download-manager[940]: tftp: server error: (1) File not found
Sep 02 15:13:26 witherspoon phosphor-version-software-manager[1052]: Error tarball does not exist
Sep 02 15:13:26 witherspoon phosphor-version-software-manager[1052]: An error when reading the Manifest file.
Sep 02 15:13:26 witherspoon dbus[943]: [system] Activating via systemd: service name='org.freedesktop.journal1' unit='dbus-org.freedesktop.journal1.service'
Sep 02 15:13:26 witherspoon dbus[943]: [system] Successfully activated service 'org.freedesktop.journal1'
Sep 02 15:13:26 witherspoon phosphor-fru-fault-monitor[1152]: /xyz/openbmc_project/logging/entry/4
Sep 02 15:13:26 witherspoon phosphor-version-software-manager[1052]: Error processing image
```

amboar, 2017-08-31:
As a note, running `tftp -g -r witherspoon-sequoia.pnor.squashfs.tar <IP>` manually on the box succeeds.

bradbishop, 2018-04-03:
For http uploads we have:

/upload/image

One option would be adding a new endpoint for tftp:

/upload/tftp

Where the post data would be the information necessary for the route handler to call the download-tftp dbus interface.

From there we can use the same code to block the route handler and return the new software id that the upload/image route handler uses.

anoo1, 2018-08-01:
Going to close this issue because it cannot be determined how long a TFTP upload can take.

A TFTP upload from a server in the same network takes over a min to complete. Making changes to add a tftp upload to the REST server already causes an nginx timeout. And even if we increase that timeout, an upload from a remote server can take a longer amount that could be anything, so we'd just be guessing when increasing timeouts.

```
curl b cjar -k -H "Content-Type: application/json" -X POST     -d '{"data": ["obmc-phosphor-image-witherspoon.ubi.mtd.tar", "x.x.x.x"]}'     https://${bmc}/xyz/openbmc_project/software/action/DownloadViaTFTP
<html>
<head><title>504 Gateway Time-out</title></head>
<body bgcolor="white">
<center><h1>504 Gateway Time-out</h1></center>
<hr><center>nginx</center>
</body>
</html>
```

closed 2018-08-01 by anoo1