openbmc/openbmc #1692: v1.99.6 tag fails to build

shenki, 2017-05-31:
On an Ubuntu x86 host:
```
$  python --version
Python 2.7.12
```

```
$ bitbake obmc-phosphor-image
NOTE: Your conf/bblayers.conf has been automatically updated.
Loading cache: 100% |##################################| Time: 0:00:00
Loaded 2642 entries from dependency cache.
ERROR: /scratch/joel/openbmc/meta/recipes-core/os-release/os-release.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 12, function: <module>
     0008:__anon_35__scratch_joel_openbmc_meta_classes_devshell_bbclass(d)
     0009:__anon_108__scratch_joel_openbmc_meta_classes_sstate_bbclass(d)
     0010:__anon_45__scratch_joel_openbmc_meta_classes_blacklist_bbclass(d)
     0011:__anon_51__scratch_joel_openbmc_meta_classes_allarch_bbclass(d)
 *** 0012:__anon_20__scratch_joel_openbmc_meta_phosphor_common_recipes_core_os_release_os_release_bbappend(d)
File: '/scratch/joel/openbmc/meta-phosphor/common/recipes-core/os-release/os-release.bbappend', lineno: 15, function: __anon_20__scratch_joel_openbmc_meta_phosphor_common_recipes_core_os_release_os_release_bbappend
     0011:        if version_id:
     0012:                d.setVar('VERSION_ID', version_id)
     0013:                versionList = version_id.split('-')
     0014:                version = versionList[0] + "-" + versionList[1]
 *** 0015:                d.setVar('VERSION', version)
     0016:
     0017:        build_id = run_git(d, 'describe --abbrev=0')
     0018:        if build_id:
     0019:                d.setVar('BUILD_ID', build_id)
Exception: IndexError: list index out of range

ERROR: Failed to parse recipe: /scratch/joel/openbmc/meta/recipes-core/os-release/os-release.bb

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

shenki, 2017-05-31:
The issue is when we are on a tag the outout of `git describe` contains fewer elements:
```
$ git checkout v1.99.6
$ git describe --dirty
v1.99.6
```

```
$ git checkout master
$ git describe --dirty
v1.99.6-21-g3de8405
```

So when we split on `-` the `versionList` array is of length 1.

We could fix this by doing `git describe --dirty --long` instead.

rfrandse, 2017-06-16:
<https://gerrit.openbmc-project.xyz/4849> Fix git describe format for commits with a tag
Fixes  #1692 v1.99.6 tag fails to build

closed 2017-06-22 by williamspatrick

gtmills, 2019-09-14:
@shenki Any other reason we moved to `--long` then to fix this ?
I propose we move back to git describe --dirty  so on the web interface firmware page/overview page users will just the "tag" (assuming commit matches the tag) as the version of firmware. I think showing `-0-g<commit>` is not needed.