openbmc/openbmc #1695: Removed DD1.0 workarounds for DD2.0 support

geissonator, 2017-05-31:
When DD2.0 shows up, we need to undo the vcs workarounds and such we did for DD1.0.  This story will track that work.  It's been requested the openbmc code base support both DD1.0 and DD2.0.

geissonator, 2017-06-01:
We may possibly be able to partially power on system, read CFAM reg to get DD level, and then act accordingly, it's probably going to need some experimentation.  For this story we'll break up into a minimum  commitment, then a bonus task
Commit:
- [ ] Update workaround services to not run if dd2.0 file is present in filesystem

Bonus:
- [ ] Create new service that reads CFAM and dynamically creates DD2.0 file for above services

geissonator, 2017-06-05:
Here are the services that I will disable if dd2.0:

- ucd_disable_vcs@.service                 (Disable UCD VCS rails)
- avsbus-workaround@.service           (Perform AVS bus workaround on VRMs)
- avsbus-disable@.service                   (Disable the AVS bus on the VRMs)
- vrm-control@.service                        (Apply voltage overrides to VRMs, vdna=0.9 vdnb=0.9)
- avsbus-enable@.service                   (Enable the AVS bus on VRMs)
- vcs_off@.service                                (Disable VCS before Witherspoon power on)
(pgood asserted)
- vcs_workaround@.service                 (Run VCS workaround on host0)
- vcs_on@.service                                 (Turn VCS on)
(start sbe/host)

rfrandse, 2017-06-06:
<https://gerrit.openbmc-project.xyz/4621> Provide ability to disable DD1.0 workarounds
Resolves  #1695 Removed DD1.0 workarounds for DD2.0 support

jkeusema, 2017-06-06:
@geissonator I think you need to keep these steps 
avsbus-disable@.service (Disable the AVS bus on the VRMs)
vrm-control@.service (Apply voltage overrides to VRMs, vdna=0.9 vdnb=0.9)
avsbus-enable@.service (Enable the AVS bus on VRMs)

The voltage over-rides should just be moved back to HW defaults which is :
VDD = 1.0V
VCS = 1.03V
VDN = 1.0V

The reason we do this is to ensure that if the system was doing DVFS or setting VPD boot voltages that the voltages get reset to HW defaults before the next power-on..those 3 rails are all set either by VPD or dynamically during runtime...we don't technically have to set over-rides we can just disable AVSBus and re-enable AVSBus to get this effect but keeping the step in to perform voltage over-rides seems like a good idea and matches the way other systems are doing it.

geissonator, 2017-06-06:
ok, for now, I'll leave the avsbus-disable and avsbus-enable running on both dd1.0 and dd2.0.  I'll disable vrm-control but if someone wants to run it on dd2.0, they can simply modify the file and it will automatically run during boot.  This will ensure the VRM's get reset to their defaults properly on each boot and it's easy for someone to do some overrides if needed for testing.

rfrandse, 2017-06-08:
<https://gerrit.openbmc-project.xyz/4621> Provide ability to disable DD1.0 workarounds
Partially resolves  #1695 Removed DD1.0 workarounds for DD2.0 support

closed 2017-06-12 by williamspatrick