openbmc/openbmc #1696: Need to use correct /dev/occ* paths in occ pass through

geissonator, 2017-05-31:
Looks like there was a change in the kernel to name the /dev/ paths for occ the following:
```
ls /dev/occ
occ1  occ2  
```

Need to update openbmc/openpower-occ-control/occ_pass_through.hpp#75 with this new path

vishwabmc, 2017-06-01:
> Also, these devices only show up after we've done the initial power on and scan of the FSI. The service that starts this will need to change to only start after that. Could probably just put in obmc-host-start.target since there's no occ pass through without the host booting up and starting it.

 > This part would be handled by: https://gerrit.openbmc-project.xyz/#/c/4127/

rfrandse, 2017-06-01:
<https://gerrit.openbmc-project.xyz/4533> Use /dev/occ instead of /dev/occfifo
Fixes  #1696 Need to use correct /dev/occ* paths in occ pass through

closed 2017-06-02 by williamspatrick