openbmc/openbmc #1698: OCC hwmon driver requires a manual FSI rescan to start once OCC is active

eddiejames, 2017-05-31:
Driver is probed via FSI, but FSI starts up long before the OCC goes active. Need a service to automatically rescan FSI once the OCC is active.

eddiejames, 2017-06-16:
- [ ] Create systemd service files to bind and unbind occ-hwmon driver
- [ ] Create yaml for phosphor-dbus-monitor to watch OccStatus sensor for change in state and start appropriate service file.
- [ ] Install in OpenBmc and test it.

geissonator, 2017-06-16:
Per design discussion this morning, Vishwa will take this story and will implement the occ active sensor dbus interfaces to handle this function.

The services should basically do this I believe:
echo occ1-dev0 > /sys/bus/platform/drivers/occ-hwmon/bind
echo occ2-dev0 > /sys/bus/platform/drivers/occ-hwmon/bind

echo occ1-dev0 > /sys/bus/platform/drivers/occ-hwmon/unbind
echo occ2-dev0 > /sys/bus/platform/drivers/occ-hwmon/unbind

vishwabmc, 2017-06-20:
https://gerrit.openbmc-project.xyz/#/c/4862/

vishwabmc, 2017-06-21:
- [ ] Implement OCC Status sensor dbus objects
- [ ] Implement sd_event loop for handling user requests
- [ ] Define a class that is responsible for bind and unbind
- [ ] Define the relation between the sensor class and this bind class
- [ ] Watch for occ-error indication in the /sysfs entry
- [ ] Use Sensor class and Bind class to do the right thing when Status is set to Active and making it Inactive

rfrandse, 2017-07-01:
<https://gerrit.openbmc-project.xyz/5040> Add support to watch for OCC errors
Fixes  #1698 OCC bind and unbind as occ sensor goes active and non-active

closed 2017-07-06 by williamspatrick