openbmc/openbmc #1699: SBE/OCC on the second processor hangs

eddiejames, 2017-05-31:
Something hangs in the path (OCC hwmon -> OCC "fifo" -> SBEFIFO -> FSI) for the second processor OCC. Figure out what and fix it.

geissonator, 2017-06-15:
Fix will be included in occ upstream patches

closed 2017-06-15 by geissonator

geissonator, 2018-04-30:
Does this system have secure boot enabled @madscientist159 ?  http://patchwork.ozlabs.org/patch/905497/ was put out recently to address some issues with secure mode enabled.

madscientist159, 2018-04-30:
@geissonator No, it does not.  We are not yet shipping anything in secure mode due to various issues in the upstream firmware.

amboar, 2018-05-01:
> `[  302.890000] sbefifo 01:01:00:06: SBE SCOM access gave return code 0x0002000f, expected 0x00000000!`

None of our kernels print this message, can you point to the tree where you're seeing this error?

madscientist159, 2018-05-01:
@amboar we added that message for debugging.  Tree is here:

https://git.raptorcs.com/git/talos-obmc-linux/log/?h=dev-4.10

Patch adding the messages:
https://git.raptorcs.com/git/talos-obmc-linux/commit/?h=dev-4.10&id=ce112cef4af78a6b47ab6542602513a9a850791c

amboar, 2018-05-10:
@ozbenh is looking at the FSI stack at the moment for other reasons, he may have some insight here.

madscientist159, 2018-05-10:
@shenki Unfortunately 4.13 cripples and/or disables fan control on these platforms due to OCC instability.  Basically the OCC hwmon drivers are not only unreliable there, but the workarounds we had developed for 4.10 no longer function.  We can't ship a product that doesn't have stable fan controls due to the noise level of an uncontrolled workstation chassis.

My understanding is that there is work that needs to be done by the OCC team to make recovery of a locked up OCC / FSI bus possible without a chassis power cycle.  Is there a bug report tracking this work?