openbmc/openbmc #1701: Pinmux SPI1 error on Palmetto

shenki, 2017-06-01:
[dmesg.txt](https://github.com/openbmc/openbmc/files/1043879/dmesg.txt)
Using the patch I recently posted for dev-4.10 on Palmetto, I get this:

```
[    4.320000] pinctrl core: add 1 pinctrl maps
[    4.320000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: found group selector 122 for SPI1
[    4.570000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 68 (B22) for 1e630000.spi
[    4.570000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 69 (G19) for 1e630000.spi
[    4.570000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 70 (C18) for 1e630000.spi
[    4.570000] aspeed-g4-pinctrl 1e6e2000.syscon:pinctrl@1e6e2000: request pin 71 (E20) for 1e630000.spi
[    4.570000] Muxing pin 68 for SPI1
[    4.570000] Want SCU70[0x00003000]=0x1, got 0x2 from 0x120CE416
[    4.570000] Want SCU70[0x00003000]=0x1, got 0x2 from 0x120CE416
[    4.570000] aspeed-smc 1e630000.spi: Error applying setting, reverse things back
[    4.570000] aspeed-smc 1e630000.spi: mx25l25635e (32768 Kbytes)
```
I can still see the PNOR though:
```
# cat /proc/mtd 
dev:    size   erasesize  name
mtd0: 02000000 00010000 "bmc"
mtd1: 00060000 00010000 "u-boot"
mtd2: 00020000 00010000 "u-boot-env"
mtd3: 00440000 00010000 "kernel"
mtd4: 01740000 00010000 "rofs"
mtd5: 00400000 00010000 "rwfs"
mtd6: 02000000 00010000 "pnor"
```

shenki, 2017-06-01:
Ball | Default | Function 1 | Control 1 | Function 2 | Control 2 | Others |
| --- | ---        | ---             | ---           | ---              | ---            | ---      |
| B22 | GPIOI4 | SPICS0# | Strap[13:12]!=0 | VBCS# | Strap[5]=1 | GPIOI4 |

```
$ bitfield SCU70 0x120CE416
decoding as SCU Hardware strapping register
0x120ce416 [302834710]
                     SPI Mode Selection: 0x2
```

shenki, 2017-06-01:
**SPI mode selection**
```
00: Disable SPI interface
01: Enable SPI Master
10: Enable SPI Master and SPI Slave to AHB Bridge
11: Enable SPI Pass-through
```

So pinmux wants 0x1, but we've got 0x2. Do we fix the strapping, or fix pinmux?

shenki, 2017-06-01:
Both barreleye and palmetto set 0x2.

amboar, 2017-09-25:
Patch fixing palmetto has been included: openbmc/linux@c53f60dd34c0ad421c12d6c2d84b5f6fe063dd62

closed 2017-09-25 by rfrandse

AlooGS, 2020-07-29:
Where is  set 0x2 on barreleye and palmetto ? The FMC is used for BMC,  so I want to use the SMC on BMC system to update the host BIOS, but set SCU70[13:12] = 1 on uboot, uboot will hang :-(

AlooGS, 2020-07-29:
The OpenBMC version is 2.8.0

amboar, 2020-07-29:
@AlooGS The strapping controls the FMC as you note, you shouldn't be adjusting it to use the SMC (the SMC is unrelated).

AlooGS, 2020-07-29:
But the kernel pinctl driver to judge SCU70[13:12] , and my SCU70[13:12] = 0, so the SMC driver is not worked. so i think SCU[13:12] is useful for SMC, my machine is AST2400.

AlooGS, 2020-07-29:
Sorry, it is not SMC, the SPI Flash contoller  is 0x1E630000.

AlooGS, 2020-07-29:
[image]
I want to use the blue.

amboar, 2020-07-29:
Ah, sorry, it's been a while - you're right, these bits are associated with the SMC (BIOS SPI flash controller).

The pinctrl driver treats the strapping (SCU70) as read-only for the most part. It assumes you've strapped the BMC correctly with respect to your system design. If you need `SCU[13:12] = 0b10` then you should have a pull-up on ball L19 (ROMA13 / GPIOAA3) and ball L18 (ROMA12 / GPIOAA2) should be left floating.

We could consider making the pinctrl driver consider these bits read/write, but ideally the strapping should be correct in hardware. If you just want to do a quick test you can use `devmem 0x1e6e2070 ...` from Linux userspace.

AlooGS, 2020-07-29:
yes, linux userspace or kernelspace is can test, but uboot do not , i think do it by uboot is very well.
 

王遥
lemote

 联系我

 
 
 
------------------ Original ------------------
From: "Andrew Jeffery"; 
Date: 2020年7月29日(星期三) 中午11:14
To: "openbmc/openbmc"; 
Cc: "Yao Wang"; "Mention"; 
Subject: Re: [openbmc/openbmc] Pinmux SPI1 error on Palmetto (#1701)

 

 
Ah, sorry, it's been a while - you're right, these bits are associated with the SMC (BIOS SPI flash controller).
 
The pinctrl driver treats the strapping (SCU70) as read-only for the most part. It assumes you've strapped the BMC correctly with respect to your system design. If you need SCU[13:12] = 0b10 then you should have a pull-up on ball L19 (ROMA13 / GPIOAA3) and ball L18 (ROMA12 / GPIOAA2) should be left floating.
 
We could consider making the pinctrl driver consider these bits read/write, but ideally the strapping should be correct in hardware. If you just want to do a quick test you can use devmem 0x1e6e2070 ... from Linux userspace.

AlooGS, 2020-07-29:
The hardware set SCU70[13:12] is not flexible than software, you know, the kernel just read , so uboot is very well.

chnguyen-ampere, 2020-12-05:
> Patch fixing palmetto has been included: [openbmc/linux@c53f60d](https://github.com/openbmc/linux/commit/c53f60dd34c0ad421c12d6c2d84b5f6fe063dd62)

I applied this patch but the SPI Flash controller is still not work.
The SCU70[13:12] still is 0x0. And I tried to set The SCU70[13:12]  to 0x1 by devmem on Linux Shell , then the SPI Flash controller work fine.

Hi @amboar @AlooGS , What's solution that you chose to set The SCU70[13:12]  to 0x1 ? In U-boot or in Kernel Linux , or anything else?

amboar, 2020-12-06:
@chnguyen-ampere the theory of pinctrl driver is that SCU70 should present the hardware design as configured by the strapping pins. It sounds like your board strapping does not reflect your intended use of the system. Again by the theory of the pinctrl driver, you should fix your board strapping to represent the intended use.

However, as it stands you can fix the board strapping up in u-boot by adding an appropriate `mw.l` to `bootcmd`.

If you have a convincing argument for allowing the kernel to change the board configuration then we can talk that over either here or on the mailing list.

chnguyen-ampere, 2020-12-07:
@amboar , my system design the strapping pins wire to the pin of another module. In some cases, the strapping pin will not reflect correctly my intended.
 In U-Boot, I support to check the SCU70 and revise if it is not matched my expectation.

amboar, 2020-12-07:
@chnguyen-ampere okay, that is an interesting use-case. Can you talk about the designs a little more? It sounds to me like you're designing a BMC card that can plug into multiple platform designs and is supported by a firmware that can adapt to the platform at runtime?

Say we change the pinctrl driver to allow selecting the SPI mode at boot time in the devicetree, do you have a strategy for selecting the right devicetree to use?