openbmc/openbmc #1702: unbind ir35221 prior to using i2c-dev

bradbishop, 2017-06-01:
The following scripts need to be updated to unbind the ir35221 driver before using i2cset:

```$ grep -rl i2cset *
usr/bin/avsbus-disable.sh
usr/bin/avsbus-workaround.sh
usr/bin/avsbus-enable.sh
usr/bin/vrm-control.sh
usr/bin/vcs_on.sh
usr/bin/ucd_disable_vcs.sh
usr/bin/vcs_off.sh

grep -rl i2cget *
usr/bin/vrm-control.sh```

bradbishop, 2017-06-01:
- [ ] identify which scripts are touching ir35221
- [ ] unbind and rebind ir35221

geissonator, 2017-06-01:
Idea here it so just bind/unbind in the services that need the access

amboar, 2017-06-02:
Was there an issue with the `i2c{g,s}et -f` (force) option?

cbostic, 2017-06-02:
The force option was discussed in today's scrum.  Given that the ir35221 page must be set via an i2c write prior to other reads/writes I believe that would keep us from using -f.

cbostic, 2017-06-02:
usr/bin/ucd_disable_vcs.sh   included in the initial list of affected files is only touching i2c-11 so I assume this is not affected.  ir35221's are only on i2c-4 and i2c-5.

shenki, 2017-06-06:
The affected files appear to be this list:

- avbus-disable.sh
- vrm-control.sh
- vcs_off.sh
- avsbus-enable.sh
- vcs_on.sh

I can boot a witherspoon host after making changes to these scripts.

This task is blocking the merge of the updated kernel.

cbostic, 2017-06-06:
@shenki   My change includes those you listed plus  avsbus-workaround.sh

https://gerrit.openbmc-project.xyz/#/c/4571/

I so far have not been able to power on system,  obmcutil power indicates pgood = 0.
How does your change differ?

mine260309, 2017-06-07:
So romulus needs similar changes:
```
meta-romulus/recipes-phosphor/chassis/vrm-control/vrm-control.sh
meta-romulus/recipes-phosphor/chassis/avsbus-control/avsbus-disable.sh
meta-romulus/recipes-phosphor/chassis/avsbus-control/avsbus-enable.sh
meta-romulus/recipes-phosphor/chassis/avsbus-control/avsbus-workaround.sh
```

cbostic, 2017-06-09:
> So romulus needs similar changes:
>
> |meta-romulus/recipes-phosphor/chassis/vrm-control/vrm-control.sh
> meta-romulus/recipes-phosphor/chassis/avsbus-control/avsbus-disable.sh [...]

|Lei,

Will put that on the to do list.

Thanks
|

rfrandse, 2017-06-13:
<https://gerrit.openbmc-project.xyz/4650> Add ir35221 bind/unbind service to vrm control
Resolves  #1702 unbind ir35221 prior to using i2c-dev

closed 2017-06-14 by williamspatrick