openbmc/openbmc #1703: Failed to Upload PNOR image via REST

saqibkh, 2017-06-01:
I tried to upload a valid PNOR image tarball. The REST command returned successfully, but no tarball or image dir exists on the BMC.

FYI, I copied the same tarball on the TFTP server and tried downloading the same image via TFTP. This Passed successfully.
------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------
REST command:
$ curl -c cjar -b cjar -k -X POST -H "Content-Type: application/octet-tream" -T /esw/san5/khansa/public/w37.pnor.squashfs.tar https://9.3.23.30/upload/image
{
  "data": null,
  "message": "200 OK",
  "status": "ok"
------------------------------------------------------------------------------------------------------------------
------------------------------------------------------------------------------------------------------------------
Below is the journal log for the failed attempt which shows a failure for the validation of BMC image but not the PNOR image. 

--> journalctl -f
Apr 22 10:42:12 barreleye phosphor-gevent[1006]: ::8029:a3b6:8e91:f402%1123197948 - - [2017-04-22 10:42:12] "POST /login HTTP/1.1" 200 320 0.095012
Apr 22 10:42:52 barreleye phosphor-version-software-manager[755]: tar: short read
Apr 22 10:42:52 barreleye phosphor-version-software-manager[755]: Error No manifest file
Apr 22 10:42:52 barreleye phosphor-version-software-manager[755]: An error when reading the Manifest file.
Apr 22 10:42:52 barreleye phosphor-version-software-manager[755]: Error processing image
Apr 22 10:42:58 barreleye phosphor-version-software-manager[755]: Error tarball does not exist
Apr 22 10:42:58 barreleye phosphor-version-software-manager[755]: An error when reading the Manifest file.
Apr 22 10:42:58 barreleye phosphor-gevent[1006]: ::8029:a3b6:8e91:f402%1123197948 - - [2017-04-22 10:42:58] "POST /upload/image HTTP/1.1" 200 169 6.465830
Apr 22 10:42:58 barreleye phosphor-version-software-manager[755]: Error processing image

gkeishin, 2017-06-01:
So this is the same failure with a non valid image general file upload as well..

saqibkh, 2017-06-01:
A non-valid image will not show up in the /tmp/images dir,
while a valid one should show up. Or at the very least output some appropriate logs in the journal to reflect the error.

saqibkh, 2017-10-17:
FYI We do get an error log with invalid images.
Oct 17 14:34:19 witherspoon phosphor-version-software-manager[1234]: Error No manifest file
Oct 17 14:34:19 witherspoon phosphor-version-software-manager[1234]: An error when reading the Manifest file.

@rfrandse Please close this defect as it is no longer valid. Thanks

closed 2017-10-17 by geissonator