openbmc/openbmc #1704: Write a bbclass to simplify copying of error yaml files in native recipes

devenrao, 2017-06-01:
At present error yaml files are copied manually in the do_install method of error-native recipes. As there are many native recipes expected for the applications it is better to create a bbclass that takes in parameters and automates the copying of error yaml files.

Refer to openpower-debug-collector-error-native.bb how it is done manually.

devenrao, 2017-06-03:
Also need to consider the review comment proposed by Brad at the below link
https://gerrit.openbmc-project.xyz/#/c/4209/15/meta-phosphor/common/recipes-phosphor/logging/phosphor-logging-error-logs-native.bb

devenrao, 2017-08-03:
# **STATUS**
03/08
Working on moving the copy function to base class, no success so far

devenrao, 2017-09-29:
No longer valid as we are using nativesdk class to upload the error yaml files through makefile changes and using BBCLASSEXTEND += native nativesdk

This story can be invalidated

gkeishin, 2017-09-29:
Closing this as request @tomjoseph83

closed 2017-09-29 by gkeishin