openbmc/openbmc #1705: Packagegroups or dependent recipes for IPMI and error native recipes

devenrao, 2017-06-01:
At present IPMI uses package group concept for the dependency issues and phosphor-logging-error-native(Issue 1449) uses recipe dependency.

Need to discuss and find a better approach.

Initially 1449 was supposed to be using package groups concept but the same has been discontinued as Brad thought they are not actually packages in Bitbake sense.

williamspatrick, 2017-06-02:
@devenrao - The referenced issue doesn't seem right.  Is it?

devenrao, 2017-09-30:
This issue can be closed as we are using native class approach for error log header generation. A new approach is followed and is implemented as part of 1772.
This can be closed or invalidated.

closed 2017-10-23 by dkodihal