openbmc/openbmc #1706: Request to add user-generated Server Name attribute

alfordja, 2017-06-01:
Sponsor users are telling us that they desire a server name or description field whose value can be entered and maintained by the user.  Some customers have standardized server naming conventions, and others simply want a memorable means of identifying specific systems.  Today in HMC, servers are identified by default using a combination of MTM and serial number, but administrators are allowed to replace the default name with a name of their choosing.  Competitive BMCs also offer user-generated name fields.  HP iLO allows up to approx. 46 characters in the Server Name field.  The current value of Server Name would be displayed in the web GUI to help users confirm that they are working with the intended server/BMC.

The default value of the Server Name field can be the Model and Serial Number fields from Inventory, or if that information is not available, blank.

A later enhancement would be to support an additional group of fields in which the administrator could enter location information, such as the grid coordinates, rack number, and U height, for a server to aid in location.

williamspatrick, 2017-06-01:
This is fairly trivial.  Assigning to @rfrandse to prioritize and assign into a sprint.

williamspatrick, 2017-06-02:
My suggestion would be:

1. create a new dbus interface to hold this.  Maybe @alfordja can recommend a good name.  I think some other interfaces call this an 'Asset Tag'.

2. Add the interface to the phosphor-settings-manager YAML that @dkodihal recently implemented.

3. Add a bitbake enhancement to default the setting (via a YAML append) to the machine name (ie. witherspoon, zaius)?

gtmills, 2018-05-21:
@alfordja Is this still needed? 

The system inventory item inherits the AssetTag, https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/Inventory/Decorator/AssetTag.interface.yaml, interface.

```
 curl -b cjar -k https://$BMC_IP/xyz/openbmc_project/inventory/system
{
  "data": {
    "AssetTag": "", 
    "BuildDate": "", 
    "Cached": 0, 
    "FieldReplaceable": 0, 
    "Manufacturer": "", 
    "Model": "8335-GTG        ", 
    "PartNumber": "", 
    "Present": 1, 
    "PrettyName": "", 
    "SerialNumber": "WITHERSPOON00055"
  }, 
  "message": "200 OK", 
  "status": "ok"
}
```

Looks like we aren't setting it though on Witherspoon. Should we be?

dkodihal, 2018-05-22:
@gtmills The AssetTag property can be set and maintained by a user. We don't set it to an initial default though.

gtmills, 2018-07-09:
@alfordja Is being able to set `AssetTag` good enough? 
Can this be closed?

closed 2019-10-27 by stale[bot]