openbmc/openbmc #1709: Inventory API enumerated data has empty fields and/or cryptic values

whowutwut, 2017-06-01:
Some general comments while looking at the Inventory Endpoint that I thought I would bring up to either verify that it's correct or point out things missing.  

This following data is collected when running the following versions: 
```
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-10"
[                                                  ] 0%IBM-witherspoon-ibm-OP9_v1.16_1.24
```

There's a test property that is still being returned in inventory that I think should be removed: 
```
    "/xyz/openbmc_project/inventory/createme3": {
      "ExampleProperty1": "foo"
    }
```

### Pretty Name:
* `chassis/motherboard/cpu/coreXX` - all the PrettyName fields are empty
* `chassis/motherboard/dimmXX` - Pretty name is : 0x0c ,  not too pretty....   :-) 
* `system` - Pretty name is empty 

### Build Date: 
* `chassis/motherboard/dimmX` - Build Date Fields are all empty
* `chassis/motherboard/cpuX` - Build Date fields are empty

### Part Number 
* `chassis/motherboard/dimmX` - Part Number fields are all empty 

### Model 
* `system` - value for Model=2, what does this mean? 
* `chassis/motherboard/cpuX` - Model value is empty 

### Manufacturer 
* `system` - Manufacturer is also empty 
* `system/chassis/motherboard` is all 0s, maybe this is just our EUH machine issue. 
* `system/chassis/motherboard/dimmXX` - Manufacturer is 0xce80 - who is this?  

Is there a mapping between those hex values and vendor names?

williamspatrick, 2017-06-01:
Most of these come to us from the Host firmware and either they are not providing it or they are providing it in the ugly form.  I'm going to assign @geissonator for now to interlock with the Host firmware on where they are with providing some more of this information and cleaning it up.

I think that on some of them, such as the mfg (0xce80), ipmitool has a giant table built in that translates between JEDEC-assigned mfg numbers and something human readable.  We do not have that mapping natively in the BMC today.

@ratagupt - FYI also.  I think some of these you were already tracking down in an issue opened by the Host firmware team.

williamspatrick, 2017-06-01:
@geissonator @rfrandse - I listed this as an enhancment.  I expect this will become an epic and stories will be created under this.

whowutwut, 2017-06-19:
Could we bubble up the issue to remove this from the return?
```
   "/xyz/openbmc_project/inventory/createme3": {
      "ExampleProperty1": "foo"
    }
```
Or is that kept in the return for some specific reason?   because it doesn't have the "Present=1" field, we are flagging this return data as an error.

geissonator, 2017-07-13:
#1357 in this area as well.

geissonator, 2017-07-29:
# **STATUS**
- No progress yet, goal here is to review anything not covered under other issues required for beta.

geissonator, 2017-08-07:
Using our latest code, I did a boot, here are all errors from ipmid:
```
ERROR Unable to get the fru info for FRU=20
Error updating inventory
ERROR Unable to get the fru info for FRU=26
Error updating inventory
ERROR Unable to get the fru info for FRU=26
Error updating inventory
ERROR Unable to get the fru info for FRU=26
Error updating inventory
ERROR Unable to get the fru info for FRU=26
Error updating inventory
ERROR Unable to get the fru info for FRU=26
Error updating inventory
ERROR:[0x80] handling NetFn:[0x2C], Cmd:[0x3]
Failed to create a method call: Invalid argumentBus: :1.3 Path:  Interface: org.openbmc.InventoryItem 
Failed to get a fru type: Invalid argumentBus: :1.3, Path: , Interface: org.openbmc.InventoryItem
ERROR:[0xCB] handling NetFn:[0x4], Cmd:[0x2D]
Failed to get /org/openbmc/inventory/system/powerlimit busname: No such file or directory
Could not find sensor 8
Failed to create a method call: Invalid argumentBus: :1.3 Path:  Interface: org.openbmc.InventoryItem 
Failed to get a fru type: Invalid argumentNo registered handlers for NetFn:[0x3A], Cmd:[0x1A] trying Wilcard implementation 
No Registered handlers for NetFn:[0x3A],Cmd:[0xFF]
ERROR:[0xC1] handling NetFn:[0x3A], Cmd:[0x1A]
Failed to create a method call: Invalid argumentBus: :1.3 Path:  Interface: org.openbmc.InventoryItem 
Failed to get a fru type: Invalid argument
```

geissonator, 2017-08-09:
Talked with Victor, nothing critical here for beta here so moving out.

whowutwut, 2017-08-15:
@bradbishop @geissonator 

In the xCAT code, I flagged an ERROR out when we receive what we thought was incomplete data in the REST calls.    The only check that we have done is if the "Present" Value is not there... 

```
            if (! defined $content{Present}) {
                # This should never happen, but if we find this, contact firmware team to fix...
                xCAT::SvrUtils::sendmsg("ERROR: Invalid data for $key_url, contact firmware team!", $callback, $node);
                next; 
}
```

@ralphbellofatto  brought this up on the xCAT issue mentioned above....  running 1730C firmware, on his nodes, results in the following: 
```
c460c802: ERROR: Invalid data for /xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/fault, contact firmware team!
c460c802: ERROR: Invalid data for /xyz/openbmc_project/inventory/system/chassis/motherboard/cpu1/fault, contact firmware team!
c460c802: ERROR: Invalid data for /xyz/openbmc_project/inventory/system/fault, contact firmware team!
c460c802: ERROR: Invalid data for /xyz/openbmc_project/inventory/system/chassis, contact firmware team!
```

I looked into this...
```
    "/xyz/openbmc_project/inventory/system/chassis/motherboard/cpu0/fault": {
      "endpoints": [
        "/xyz/openbmc_project/logging/entry/20",
        "/xyz/openbmc_project/logging/entry/32",
        "/xyz/openbmc_project/logging/entry/27",
        "/xyz/openbmc_project/logging/entry/24",
        "/xyz/openbmc_project/logging/entry/36",
        "/xyz/openbmc_project/logging/entry/39",
        "/xyz/openbmc_project/logging/entry/11",
        "/xyz/openbmc_project/logging/entry/17"
      ]
    },

    "/xyz/openbmc_project/inventory/system/chassis": {
      "AirCooled": 1,
      "WaterCooled": 0
    },

    "/xyz/openbmc_project/inventory/system/fault": {
      "endpoints": [
        "/xyz/openbmc_project/logging/entry/22",
        "/xyz/openbmc_project/logging/entry/23",
        "/xyz/openbmc_project/logging/entry/26",
        "/xyz/openbmc_project/logging/entry/41",
        "/xyz/openbmc_project/logging/entry/42",
        "/xyz/openbmc_project/logging/entry/29",
        "/xyz/openbmc_project/logging/entry/9",
        "/xyz/openbmc_project/logging/entry/8",
        "/xyz/openbmc_project/logging/entry/3",
        "/xyz/openbmc_project/logging/entry/2",
        "/xyz/openbmc_project/logging/entry/4",
        "/xyz/openbmc_project/logging/entry/31",
        "/xyz/openbmc_project/logging/entry/30",
        "/xyz/openbmc_project/logging/entry/35",
        "/xyz/openbmc_project/logging/entry/34",
        "/xyz/openbmc_project/logging/entry/19",
        "/xyz/openbmc_project/logging/entry/38",
        "/xyz/openbmc_project/logging/entry/15",
        "/xyz/openbmc_project/logging/entry/14",
        "/xyz/openbmc_project/logging/entry/13",
        "/xyz/openbmc_project/logging/entry/10",
        "/xyz/openbmc_project/logging/entry/16"
      ]
    }
  }
```

So looks like there is new data and it doesn't have the Present field.  

The purpose of this check was to see if there was anything returned that looked malformed and on earlier calls we agreed that maybe it was valuable.   Is this check needed anymore?  I can't tell if this is programmed correctly or missing fields...  `Present` doesn't seem to apply to those new fields.

geissonator, 2017-08-16:
@dkodihal - I believe these are just associations? Showing that the error logs have an association to those callouts (cpu and system).

whowutwut, 2017-08-16:
Maybe the path we should take is only print the inventory attributes that have the `Present` field in the attribute, and ignore the other ones.

dkodihal, 2017-08-16:
Right, these are error associations - a logical concept, but pointing to a real error. Not printing these would cause not showing things like callouts.

geissonator, 2017-10-17:
Seems like we're good here, no plans to make any code changes with this one.

closed 2017-10-17 by geissonator