openbmc/openbmc #1710: Mapper response empty on some path/interfaces

msbarth, 2017-06-01:
During fan degrade testing, found that the mapper returns an empty response for the same inventory path but different interfaces while attempting to get the inventory manager service to read a property.

Getting the inventory manager's service for the following was successful:
  path = /xyz/openbmc_project/inventory/system/chassis/motherboard/fan0
  intf = xyz.openbmc_project.Inventory.Item

However, the following attempt to get the inventory manager's service failed:
  path = /xyz/openbmc_project/inventory/system/chassis/motherboard/fan0
  intf = xyz.openbmc_project.State.Decorator.OperationalStatus

This was produced on v1.99.6 using the openbmc/phosphor-fan-presence/utility.cpp::getService function

bradbishop, 2017-06-02:
I would guess that its PIM rather, that is not adding the OperationalStatus to the fan.

gtmills, 2017-06-06:
https://gerrit.openbmc-project.xyz/#/c/4622/

rfrandse, 2017-06-06:
<https://gerrit.openbmc-project.xyz/4622> Emit an interfaces added signal
Resolves  #1710 Mapper response empty on some path/interfaces

closed 2017-06-07 by williamspatrick