openbmc/openbmc #1711: busybox watchdog doesn't restart WDT on app restart

spinler, 2017-06-01:
I'm testing the use of the watchdog program provided by busybox to control the fan watchdog.  (Full disclosure: The busybox.cfg change to enable the watchdog isn't merged yet.)

The usecase is when a fan related process fails, we will give the watchdog process a SIGKILL so it stops pinging WDT1.

When this happens, and then when we restart the watchdog again, which we could do on the next power cycle, I can see WDT1 never actually restarts.

The following shows this:

Watchdog is running:
```root@witherspoon:~# ps |grep watchdog0
  987 root       0:00 /sbin/watchdog -T 3 -t 1 -F /dev/watchdog0

root@witherspoon:~# devmem 0x1e78500c
0x00000013
```

Manually set the watchdog to pulse a GPIO instead of resetting the system, since the driver API to do this isn't merged yet:

```root@witherspoon:~#  devmem 0x1e78500c 32 0x00000019```

Kill the watchdog:
```
root@witherspoon:~# kill -KILL 987
Jun 01 17:39:31 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!

root@witherspoon:~# devmem 0x1e78500c
0x00000018  #not running anymore
root@witherspoon:~# devmem 0x1e785010
0x00000401  #timeout bit is on
```

So far so good.  Now, restart it and check the registers:
```
root@witherspoon:~# /sbin/watchdog -T 3 -t 1 -F /dev/watchdog0 &
[2] 1247
root@witherspoon:~# ps | grep watchdog0
 1247 root       0:00 /sbin/watchdog -T 3 -t 1 -F /dev/watchdog0

root@witherspoon:~# devmem 0x1e78500c
0x00000018   #not running
root@witherspoon:~# devmem 0x1e785000
0x002DC6C0  #counter not counting down
root@witherspoon:~# devmem 0x1e785000 
0x002DC6C0
```

I imagine there may be some cleanup that needs to be done?  This program was designed to never stop running, and if it ever did a reboot would occur anyway.

williamspatrick, 2017-06-02:
Well, if the BMC dies we also want the WDT to fire.

amboar, 2017-06-02:
Fair enough.

spinler, 2017-06-02:
This probably goes away if I properly stop the watchdog before restarting it:
```echo V > /dev/watchdog0```

spinler, 2017-06-02:
I just need to do this somewhere.  closing.

closed 2017-06-02 by spinler