openbmc/openbmc #1713: Failed to start OpenPower Software Update Manager [ Witherspoon ]

gkeishin, 2017-06-02:
Driver Info
```
- [x] BUILD ID   : https://openpower.xyz/job/openbmc-build/777/
- [x] VERSION_ID : v1.99.6-31-g792087b
- [x] Kernel     : Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Fri Jun 2 04:20:38 UTC 2017 armv6l GNU/Linux
- [x] PNOR VERSION:

IBM-witherspoon-ibm-OP9_v1.16_1.24
        op-build-v1.16-83-ge67b53c-dirty
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-rc2
        hostboot-d02273f
        linux-4.11.1-openpower1-pc11ec03
        petitboot-v1.4.2-pcb6424c
        machine-xml-e536539
        occ-8a7df03
        hostboot-binaries-d8f4ab7
        capp-ucode-9c73e9f
        sbe-d770027
```

Error snippet
```

Apr 20 17:29:40 witherspoon systemd[857]: org.open_power.Software.Host.Updater.service: Failed at step EXEC spawning /usr/sbin/openpower-update-manager: No such file or directory
Apr 20 17:29:40 witherspoon systemd[1]: Started Phosphor Console Muxer listening on device /dev/ttyVUART0.
Apr 20 17:29:40 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Main process exited, code=exited, status=203/EXEC
Apr 20 17:29:40 witherspoon systemd[1]: Failed to start OpenPower Software Update Manager.
Apr 20 17:29:40 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Unit entered failed state.
Apr 20 17:29:40 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Failed with result 'exit-code'.
Apr 20 17:29:40 witherspoon systemd[1]: Starting Phosphor Version Software Manager...
Apr 20 17:29:40 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Service hold-off time over, scheduling restart.
Apr 20 17:29:40 witherspoon systemd[1]: Stopped OpenPower Software Update Manager.
Apr 20 17:29:40 witherspoon systemd[1]: Starting OpenPower Software Update Manager...

● org.open_power.Software.Host.Updater.service - OpenPower Software Update Manager
   Loaded: loaded (/lib/systemd/system/org.open_power.Software.Host.Updater.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2017-04-20 17:29:43 UTC; 7min ago
  Process: 872 ExecStart=/usr/sbin/openpower-update-manager (code=exited, status=203/EXEC)
 Main PID: 872 (code=exited, status=203/EXEC)

Apr 20 17:29:42 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Main process exited, code=exited, status=203/EXEC
Apr 20 17:29:42 witherspoon systemd[1]: Failed to start OpenPower Software Update Manager.
Apr 20 17:29:42 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Unit entered failed state.
Apr 20 17:29:42 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Failed with result 'exit-code'.
Apr 20 17:29:43 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Service hold-off time over, scheduling restart.
Apr 20 17:29:43 witherspoon systemd[1]: Stopped OpenPower Software Update Manager.
Apr 20 17:29:43 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Start request repeated too quickly.
Apr 20 17:29:43 witherspoon systemd[1]: Failed to start OpenPower Software Update Manager.
Apr 20 17:29:43 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Unit entered failed state.
Apr 20 17:29:43 witherspoon systemd[1]: org.open_power.Software.Host.Updater.service: Failed with result 'exit-code'.
root@witherspoon:~#
```

- [x] Journald log 
[journald_v1.99.6-31-g792087b_software_updater.txt](https://github.com/openbmc/openbmc/files/1046952/journald_v1.99.6-31-g792087b_software_updater.txt)

anoo1, 2017-06-02:
This was introduced with commit 69d3de0 from yesterday.
Fixing via: https://gerrit.openbmc-project.xyz/#/c/4560/

closed 2017-06-03 by williamspatrick

gkeishin, 2017-06-05:
Verified fix in latest build https://openpower.xyz/job/openbmc-build/788/

```
v1.99.6-59-gbdfe295
```