openbmc/openbmc #1718: REST upload failed : IOError: [Errno 28] No space left on device

gkeishin, 2017-06-05:
@anoo1 This one is with the 4.10 kernel and with the software manager managing the uploads image deletion.. So looks like here also even after the manager deletes it, the tmpfs is growing with every upload.

```
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-59"
VERSION_ID="v1.99.6-59-gbdfe295"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-59"
BUILD_ID="v1.99.6"
root@witherspoon:~#
 
root@witherspoon:~# uname -a
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Sun Jun 4 11:23:35 UTC 2017 armv6l GNU/Linux
```

**Before test**

```
root@witherspoon:~# df -h
Filesystem                Size      Used Available Use% Mounted on
dev                     210.1M         0    210.1M   0% /dev
tmpfs                   211.4M     13.9M    197.4M   7% /run
/dev/mtdblock4           13.3M     13.3M         0 100% /run/initramfs/ro
/dev/mtdblock5            4.0M    336.0K      3.7M   8% /run/initramfs/rw
cow                       4.0M    336.0K      3.7M   8% /
tmpfs                   211.4M         0    211.4M   0% /dev/shm
tmpfs                   211.4M         0    211.4M   0% /sys/fs/cgroup
tmpfs                   211.4M         0    211.4M   0% /tmp
tmpfs                   211.4M     60.0K    211.3M   0% /var/volatile
root@witherspoon:~#
```

**After test**
```
root@witherspoon:~# df -h
Filesystem                Size      Used Available Use% Mounted on
dev                     210.1M         0    210.1M   0% /dev
tmpfs                   211.4M     13.9M    197.4M   7% /run
/dev/mtdblock4           13.3M     13.3M         0 100% /run/initramfs/ro
/dev/mtdblock5            4.0M    336.0K      3.7M   8% /run/initramfs/rw
cow                       4.0M    336.0K      3.7M   8% /
tmpfs                   211.4M         0    211.4M   0% /dev/shm
tmpfs                   211.4M         0    211.4M   0% /sys/fs/cgroup
tmpfs                   211.4M    183.2M     28.2M  87% /tmp
tmpfs                   211.4M    384.0K    211.0M   0% /var/volatile
root@witherspoon:~#

root@witherspoon:~# ls -lart /tmp/images/
total 0
drwxrwxrwt    9 root     root           600 Apr 20 17:58 ..
drwxr-xr-x    2 root     root            40 Apr 20 17:58 .
root@witherspoon:~#

root@witherspoon:~# du -a /tmp
4       /tmp/ipmifru1a
4       /tmp/ipmifru13
4       /tmp/ipmifru12
4       /tmp/ipmifru11
4       /tmp/ipmifru10
4       /tmp/ipmifru0f
4       /tmp/ipmifru0e
4       /tmp/ipmifru0d
4       /tmp/ipmifru0c
4       /tmp/ipmifru0b
4       /tmp/ipmifru0a
4       /tmp/ipmifru09
4       /tmp/ipmifru08
4       /tmp/ipmifru07
4       /tmp/ipmifru06
4       /tmp/ipmifru05
4       /tmp/ipmifru04
4       /tmp/ipmifru03
4       /tmp/ipmifru02
4       /tmp/ipmifru01
4       /tmp/esel
0       /tmp/images
0       /tmp/systemd-private-294e4566649444ed949b8e02a1a91c12-systemd-timesyncd.service-AeioNQ/tmp
0       /tmp/systemd-private-294e4566649444ed949b8e02a1a91c12-systemd-timesyncd.service-AeioNQ
0       /tmp/.Test-unix
0       /tmp/.font-unix
0       /tmp/.XIM-unix
0       /tmp/.ICE-unix
0       /tmp/.X11-unix
84      /tmp
root@witherspoon:~#
```

```
Apr 20 17:58:15 witherspoon phosphor-gevent[988]: ::6099:6376:8e91:f402%1995348764 - - [2017-04-20 17:58:15] "POST /login HTTP/1.1" 200 320 0.033833
Apr 20 17:58:22 witherspoon phosphor-version-software-manager[853]: tar: short read
Apr 20 17:58:22 witherspoon phosphor-version-software-manager[853]: Error No manifest file
Apr 20 17:58:22 witherspoon phosphor-version-software-manager[853]: An error when reading the Manifest file.
Apr 20 17:58:22 witherspoon phosphor-version-software-manager[853]: Error processing image
Apr 20 17:58:33 witherspoon phosphor-version-software-manager[853]: Error tarball does not exist
Apr 20 17:58:33 witherspoon phosphor-version-software-manager[853]: An error when reading the Manifest file.
Apr 20 17:58:33 witherspoon phosphor-version-software-manager[853]: Error processing image
Apr 20 17:58:33 witherspoon phosphor-gevent[988]: Traceback (most recent call last):
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 862, in _handle
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     return route.call(**args)
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 619, in __call__
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     return self.callback(*a, **kw)
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 676, in wrap
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     return callback(*a, **kw)
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 854, in __call__
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     return self.callback(*a, **kw)
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 828, in wrap
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     return self.to_jsonp(callback(*a, **kw))
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/bottle.py", line 1734, in wrapper
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     rv = callback(*a, **ka)
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 793, in wrap
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     data = callback(*a, **kw)
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 103, in __call__
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     return getattr(self, 'do_' + request.method.lower())(**kw)
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 570, in do_post
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     ImageUploadUtils.do_upload()
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:   File "/usr/lib/python2.7/site-packages/obmc/wsgi/apps/rest_dbus.py", line 555, in do_upload
Apr 20 17:58:33 witherspoon phosphor-gevent[988]:     fd.write(request.body.read())
Apr 20 17:58:33 witherspoon phosphor-gevent[988]: IOError: [Errno 28] No space left on device
```

- [x] Journal log 
[journald_v1.99.6-59-gbdfe295_tmpfs_full.txt](https://github.com/openbmc/openbmc/files/1051735/journald_v1.99.6-59-gbdfe295_tmpfs_full.txt)

lgon, 2017-06-07:
Restarting the phosphor-gevent.service causes df to once again show the correct size used for /tmp.
`root@qemuarm:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                59365     44448     10331  81% /
devtmpfs                124196         0    124196   0% /dev
tmpfs                   124404         0    124404   0% /dev/shm
tmpfs                   124404      8544    115860   7% /run
tmpfs                   124404         0    124404   0% /sys/fs/cgroup
tmpfs                   124404      8032    116372   6% /tmp
tmpfs                   124404       248    124156   0% /var/volatile
root@qemuarm:~# ls -la /tmp                  
total 1
drwxrwxrwt    9 root     root           200 Jun  7 20:35 .
drwxr-xr-x   19 root     root          1024 Jun  7 14:54 ..
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .ICE-unix
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .Test-unix
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .X11-unix
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .XIM-unix
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .font-unix
-rw-------    1 root     root             0 Jun  7 20:35 .python-history
drwxr-xr-x    2 root     root            40 Jun  7 20:35 images
drwx------    3 root     root            60 Jun  7 20:27 systemd-private-797ddc21463c4dbca17d3822077fca45-systemd-timesyncd.service-ear7R8
root@qemuarm:~# systemctl restart phosphor-gevent.service
root@qemuarm:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                59365     44448     10331  81% /
devtmpfs                124196         0    124196   0% /dev
tmpfs                   124404         0    124404   0% /dev/shm
tmpfs                   124404      8544    115860   7% /run
tmpfs                   124404         0    124404   0% /sys/fs/cgroup
tmpfs                   124404         0    124404   0% /tmp
tmpfs                   124404       280    124124   0% /var/volatile
root@qemuarm:~# ls -la /tmp
total 1
drwxrwxrwt    9 root     root           200 Jun  7 20:36 .
drwxr-xr-x   19 root     root          1024 Jun  7 14:54 ..
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .ICE-unix
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .Test-unix
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .X11-unix
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .XIM-unix
drwxrwxrwt    2 root     root            40 Jun  7 20:27 .font-unix
-rw-------    1 root     root             0 Jun  7 20:36 .python-history
drwxr-xr-x    2 root     root            40 Jun  7 20:35 images
drwx------    3 root     root            60 Jun  7 20:27 systemd-private-797ddc21463c4dbca17d3822077fca45-systemd-timesyncd.service-ear7R8
`

lgon, 2017-06-07:
A way to free the memory without restarting phosphor-gevent.service is to truncate the file marked for deletion. 555 is the pid of "/usr/sbin/phosphor-gevent rest_dbus"
```
root@qemuarm:/proc/555/fd# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                59365     44448     10331  81% /
devtmpfs                124196         0    124196   0% /dev
tmpfs                   124404         0    124404   0% /dev/shm
tmpfs                   124404     24928     99476  20% /run
tmpfs                   124404         0    124404   0% /sys/fs/cgroup
tmpfs                   124404      8032    116372   6% /tmp
tmpfs                   124404       228    124176   0% /var/volatile
root@qemuarm:/proc/555/fd# ls -la /proc/555/fd 
total 0
dr-x------    2 root     root             0 Jun  7 20:35 .
dr-xr-xr-x    9 root     root             0 Jun  7 20:35 ..
lr-x------    1 root     root            64 Jun  7 20:35 0 -> /dev/null
lrwx------    1 root     root            64 Jun  7 20:35 1 -> socket:[28071]
lrwx------    1 root     root            64 Jun  7 20:35 11 -> /tmp/tmp4GcGr4 (deleted)
lrwx------    1 root     root            64 Jun  7 20:35 12 -> /tmp/tmpVe2PVG (deleted)
lrwx------    1 root     root            64 Jun  7 20:35 2 -> socket:[28071]
lrwx------    1 root     root            64 Jun  7 20:35 3 -> socket:[9668]
lrwx------    1 root     root            64 Jun  7 20:35 4 -> socket:[28270]
lrwx------    1 root     root            64 Jun  7 20:35 5 -> socket:[9668]
lrwx------    1 root     root            64 Jun  7 20:35 6 -> anon_inode:[eventpoll]
lrwx------    1 root     root            64 Jun  7 20:35 7 -> anon_inode:[eventfd]
lr-x------    1 root     root            64 Jun  7 20:35 8 -> /dev/urandom
root@qemuarm:/proc/555/fd# : > /proc/555/fd/12
root@qemuarm:/proc/555/fd# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                59365     44448     10331  81% /
devtmpfs                124196         0    124196   0% /dev
tmpfs                   124404         0    124404   0% /dev/shm
tmpfs                   124404     24928     99476  20% /run
tmpfs                   124404         0    124404   0% /sys/fs/cgroup
tmpfs                   124404         0    124404   0% /tmp
tmpfs                   124404       292    124112   0% /var/volatile
```

lgon, 2017-06-08:
- [ ] Close file descriptor in rest_dbus.py to release memory in /tmp

rfrandse, 2017-06-09:
<https://gerrit.openbmc-project.xyz/4685> Close file descriptor to release memory
Resolves  #1718 REST upload failed : IOError: [Errno 28] No space left on device

closed 2017-06-13 by williamspatrick

gkeishin, 2017-06-15:
This looked good on 
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-146"
VERSION_ID="v1.99.6-146-geeae33c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-146"
BUILD_ID="v1.99.6"
root@witherspoon:~# uname -a
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Thu Jun 15 05:21:06 UTC 2017 armv6l GNU/Linux
root@witherspoon:~#

```