openbmc/openbmc #1724: Do UCD chip fault analysis

spinler, 2017-06-05:


spinler, 2017-06-06:
Also, from the power supply requirements:

 If the power-on sequence fails due to time-out or fault status (from power sequencing) and it points to the 12V/PSU then read out STATUS_WORD to look for these bits:
•	PG#
•	IOUT_OC_FAULT
•	VOUT_OV_FAULT
•	TEMPERATURE_FAULT
•	FAN_FAULT
•	MFR_SPECIFIC
If any fault bits are set then log a SEL pointing to the PSU that shows the fault bits.

spinler, 2017-06-15:
The plan is to create a new app that gets started by phosphor-gpio-monitor to do the analysis.  This will query the UCD registers and make the appropriate errors.  As the driver for the UCD doesn't support these registers yet, we'll have to stub out those calls for now.

Not sure yet what piece of code will do the power off on an error, but probably not here because UCD errors can occur that aren't fatal (don't drop SYS_PWR_OK).

spinler, 2017-06-15:
- [ ] Create a new application TBD (UCD90XXX specific?) in repository TBD  (needs design call discussion).
- [ ] Pass in the sysfs path to look at via command line, which would be env file vars in service file
- [ ] Do the analysis as described in _UCD registers and power faults openBMC.doc_, which includes reading various registers(sysfs reads probably)
- [ ] Create 'power fault' elogs, capturing registers in the elog (FFDC driven by yaml?)

geissonator, 2017-06-16:
No sizing until we determine scope (i.e. generic data driven or something more specific to UCD)

spinler, 2017-07-17:
Status:
* 7/16  On hold

rfrandse, 2017-08-22:
<https://gerrit.openbmc-project.xyz/6185> Add service to monitor the UCD90160 at runtime
Resolves  #1724 Do UCD chip fault analysis

spinler, 2017-09-05:
This isn't actually resolved.  A commit was just merged out of order.

rfrandse, 2017-09-19:
<https://gerrit.openbmc-project.xyz/6757> Revert "Revert "Add service to monitor the UCD90160 at runtime""
Resolves  #1724 Do UCD chip fault analysis

closed 2017-09-27 by geissonator