openbmc/openbmc #1725: add GPUs to inventory

spinler, 2017-06-05:


spinler, 2017-06-08:
So far, the only way we can tell which nvlink GPUs are present is by seeing that the OCC is sending down temperatures for them.   We'd have to do more negotiating with the OCC team to get them to send us their present status, which they keep internally.

geissonator, 2017-07-14:
Patrick is driving this

spinler, 2017-09-05:
For power sequencer faults, I have plans to just add the failing GPU to the error log metadata, and will work with the RAS team on the documentation to go along with it.  So I don't need the GPUs in the inventory for those.

That being said, there are new plans for hostboot to send down GPU functional sensors so they can be added to the inventory so the OCC can call them out.

geissonator, 2017-09-27:
Host is sending this down, we just need a yaml file update for it.  Functional sensor (like cores).

rfrandse, 2017-10-06:
<https://gerrit.openbmc-project.xyz/7108> Add GPU to inventory
Resolves  #1725 add GPUs to inventory

closed 2017-10-24 by geissonator