openbmc/openbmc #1726: Do GPU fault analysis to callout a GPU

spinler, 2017-06-05:


spinler, 2017-06-21:
I'm thinking we'll define the map of GPIOs to inventory objects in the same YAML that has the rest of the UCD analysis configuration.   Still need to investigate what all needs to happen to add an inventory object to an error.

spinler, 2017-08-24:
- [ ] If the UCD chip says the GPU_PGOOD or GPU_OVERTEMP summary faults are on (already being checked for), go read the 9552 that has the fault line for each specific GPU
- [ ] If a GPIO on the 9552 says a GPU is faulted, created a GPU-PGood-Fault or GPU-OverTemp error log
- [ ] Put the failing GPU in the metadata.  In the future, if GPUs are in inventory, can do a regular callout
- [ ] Only create an error for each type once for each GPU, not repeatedly
- [ ] If an OverTemp is found, shutdown the system after the error is logged (will use the existing function that also generates a SystemShutdown due to Power Fault  error log

spinler, 2017-09-05:
# **STATUS**
*

rfrandse, 2017-09-20:
<https://gerrit.openbmc-project.xyz/6768> Shutdown system on GPU over-temps
Resolves  #1726 Do GPU fault analysis to callout a GPU

closed 2017-09-26 by williamspatrick