openbmc/openbmc #1727: UCD device driver to expose registers for fault analysis

spinler, 2017-06-05:


spinler, 2017-06-20:
The chip's command reference: http://www.ti.com/lit/ug/slvu352f/slvu352f.pdf

cbostic, 2017-07-13:
Status:
Coded for ucd driver most of day and ran some limited unit tests.

cbostic, 2017-07-13:
Task list

- [ ] Identify all required sysfs files and create a directory structure
- [ ] Create a method that checks for ucd91060 and only creates files/accesses for this type only
- [ ] Code
- [ ] Unit test all files for correct I2C access of ucd90160

cbostic, 2017-07-14:
Status Update
July 14, 2017
--------------------------------------------------------------------
Yesterday
* No time spent on this story, focus was on 1783.

Today
* Not planning on working this today.  Focus is on 1636 and 1783

rfrandse, 2017-09-13:
will track this getting in upstream with 
https://github.com/openbmc/openbmc/issues/2306

closed 2017-09-13 by rfrandse