openbmc/openbmc #1728: Add GPU PGOOD fault GPIOs to device tree

spinler, 2017-06-05:


spinler, 2017-06-21:
Looks like someone else put these into the device tree for us, and it got merged into 4.10.

```	pca9552: pca9552@60 {
		compatible = "nxp,pca9552";
		reg = <0x60>;
		#address-cells = <1>;
		#size-cells = <0>;
		gpio-controller;
		#gpio-cells = <2>;

		gpio-line-names = "PS_SMBUS_RESET_N", "APSS_RESET_N",
		"GPU0_TH_OVERT_N_BUFF",	"GPU1_TH_OVERT_N_BUFF",
		"GPU2_TH_OVERT_N_BUFF", "GPU3_TH_OVERT_N_BUFF",
		"GPU4_TH_OVERT_N_BUFF",	"GPU5_TH_OVERT_N_BUFF",
		"GPU0_PWR_GOOD_BUFF", "GPU1_PWR_GOOD_BUFF",
		"GPU2_PWR_GOOD_BUFF", "GPU3_PWR_GOOD_BUFF",
		"GPU4_PWR_GOOD_BUFF", "GPU5_PWR_GOOD_BUFF",
		"12V_BREAKER_FLT_N", "THROTTLE_UNLATCHED_N";

		gpio@0 {
			reg = <0>;
			type = <PCA955X_TYPE_GPIO>;
		};
		gpio@1 {
			reg = <1>;
			type = <PCA955X_TYPE_GPIO>;
...
```

closed 2017-06-21 by spinler

bjwyman, 2017-06-23:
https://github.com/openbmc/linux/commit/db30464ee579176abc82d0fe95aa9b9ff76d0c3a#diff-aa9d99ad68a2d01d5daf50187c9f99fb