openbmc/openbmc #1730: Manually execute ipmid and use "Get Device ID" command but return zero

Kylix0713, 2017-06-06:
I try to build evb_ast2500. Use cf7e84977a51885fd3872ecbb7f621508ee9b33c commit. And I do some modification as below:

In "meta-openbmc-machines/meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf/machine/evb-ast2500.conf" add 
OBMC_MACHINE_FEATURES += " \
        obmc-phosphor-fan-mgmt \
        obmc-phosphor-chassis-mgmt \
        obmc-phosphor-flash-mgmt \
        obmc-host-ipmi \
        obmc-host-state-mgmt \
        obmc-chassis-state-mgmt \
        obmc-bmc-state-mgmt \
	phosphor-dbus-interfaces \
	"

PREFERRED_PROVIDER_virtual/obmc-host-ipmi-hw  = "phosphor-ipmi-hw-example"
PREFERRED_PROVIDER_virtual/obmc-chassis-mgmt = "obmc-phosphor-chassisd"
PREFERRED_PROVIDER_virtual/obmc-fan-mgmt = "obmc-phosphor-fand"
PREFERRED_PROVIDER_virtual/obmc-flash-mgmt = "obmc-phosphor-flashd"
PREFERRED_PROVIDER_virtual/obmc-system-mgmt = "obmc-phosphor-sysd"
PREFERRED_PROVIDER_virtual/obmc-inventory-data = "obmc-inventory"
PREFERRED_PROVIDER_virtual/phosphor-ipmi-fru-hostfw-config = "phosphor-ipmi-fru-hostfw-config-example-native"

OBMC_IMAGE_EXTRA_INSTALL_append = " phosphor-state-manager-discover"

==============================================================================

After booting and logging in, I find that ipmid will not be executed automatically. Execute ipmid manually but get the messages as below:

root@evb-ast2500:\~# ipmid &
[1] 2849
root@evb-ast2500:\~# Registering handler:[/usr/lib/host-ipmid/libsysintfcmds.so.0]
Registering NetFn:[0x6], Cmd:[0x35]
Registering NetFn:[0x6], Cmd:[0x2E]
Registering NetFn:[0x6], Cmd:[0x31]
Registering handler:[/usr/lib/host-ipmid/libstrgfnhandler.so.1]
Registering NetFn:[0xA], Cmd:[0x12]
Registering handler:[/usr/lib/host-ipmid/libapphandler.so.0]
Registering NetFn:[0x6], Cmd:[0x36]
Registering NetFn:[0x6], Cmd:[0xFF]
Registering NetFn:[0x6], Cmd:[0x22]
Registering NetFn:[0x6], Cmd:[0x24]
Registering NetFn:[0x6], Cmd:[0x1]
Registering NetFn:[0x6], Cmd:[0x4]
Registering NetFn:[0x6], Cmd:[0x8]
Registering NetFn:[0x6], Cmd:[0x6]
Registering NetFn:[0x6], Cmd:[0x40]
Registering NetFn:[0x6], Cmd:[0x42]
Registering NetFn:[0x4], Cmd:[0xFF]
Registering NetFn:[0x4], Cmd:[0x2F]
Registering NetFn:[0x4], Cmd:[0x30]
Registering NetFn:[0x4], Cmd:[0x2D]
Registering NetFn:[0xA], Cmd:[0xFF]
Registering NetFn:[0xA], Cmd:[0x48]
Registering NetFn:[0xA], Cmd:[0x49]
Registering NetFn:[0xA], Cmd:[0x40]
Registering NetFn:[0xA], Cmd:[0x42]
Registering NetFn:[0xA], Cmd:[0x44]
Registering NetFn:[0x0], Cmd:[0xFF]
Registering NetFn:[0x0], Cmd:[0x0]
Registering NetFn:[0x0], Cmd:[0x9]
Registering NetFn:[0x0], Cmd:[0x1]
Registering NetFn:[0x0], Cmd:[0x2]
Registering NetFn:[0x0], Cmd:[0x8]
Registering NetFn:[0x2C], Cmd:[0x3]
Registering NetFn:[0xC], Cmd:[0xFF]
Registering NetFn:[0xC], Cmd:[0x1]
Registering NetFn:[0xC], Cmd:[0x2]
Registering NetFn:[0x6], Cmd:[0x2]
Registering NetFn:[0x6], Cmd:[0x3]
Registering NetFn:[0x2C], Cmd:[0x0]
"Failed to get /org/openbmc/settings/host0 busname: No such file or directory"

==============================================================================

After the above, use "ipmitool -I lanplus -H 192.168.11.6 -P 0penBmc raw 0x06 0x01", and then return "00 00 00 00 00 00 00 00 00 00 00 00 00 00 00". Does this result result from the error message of "Fail to get ..../host0"? Could someone help me get know of the reason of the failure?

gtmills, 2018-05-31:
@Kylix0713 Still a problem or can this be closed?

gtmills, 2018-08-20:
@Kylix0713 Can you close? Trying to clean up old issues. :) thanks!