openbmc/openbmc #1731: Monitor for faults at standby

spinler, 2017-06-06:


bjwyman, 2017-07-10:
[ ] Create a power supply monitoring application (phosphor-power-psu?)
[ ] Monitor STATUS_WORD and VIN_UV_FAULT values, create error log if bad values.
[ ] Update recipe and service files to start monitor application at standby

bjwyman, 2017-07-13:
STATUS
07/13: 
* On hold

bjwyman, 2017-07-14:
Final/result and testing will depend on openbmc/openbmc#1883.

williamspatrick, 2017-07-27:
Cannot commit to sprint 17.07.26 because of missing dependency on device driver.

rfrandse, 2017-08-23:
<https://gerrit.openbmc-project.xyz/6018> Power supply monitor recipe, service, and conf
Resolves:  #1731 Monitor for faults at standby

bjwyman, 2017-09-27:
This probably will not work good. @bradbishop @geissonator 
https://gerrit.openbmc-project.xyz/#/c/6018/
That depends on the updates in the witherspoon-pfault-analysis repo, which did not get merged.

Depends-On: Ie364a3dc5b7f19cbc587450fc54f231d1ec280f7

These need +2 and merge:
https://gerrit.openbmc-project.xyz/5527
https://gerrit.openbmc-project.xyz/5714
https://gerrit.openbmc-project.xyz/5861
https://gerrit.openbmc-project.xyz/5917
https://gerrit.openbmc-project.xyz/5961
https://gerrit.openbmc-project.xyz/6047
https://gerrit.openbmc-project.xyz/6048

spinler, 2017-09-27:
Not actually complete.

spinler, 2017-10-02:
Now complete.

closed 2017-10-02 by spinler