openbmc/openbmc #1732: Monitor for faults at runtime

spinler, 2017-06-06:


bjwyman, 2017-08-10:
Task List 
- [ ] Update standby fault monitor to detect power on state (openbmc/openbmc#1731)
- [ ] If power on fails due to time out or fault status, check STATUS_WORD for PG#, IOUT_OC_FAULT, VOUT_OV_FAULT, TEMPERATURE_FAULT, FAN_FAULT, or MFR_SPECIFIC. Log SEL pointing to the power supply that shows any of those fault bits.
- [ ] After IPL complete call out power supply if PG# or UNIT_IS_OFF shows fault.
- [ ] After IPL complete call out power supply if IOUT_OC_FAULT (overcurrent fault seen).
- [ ] After IPL complete call out power supply if VOUT_OV_FAULT (output over voltage fault).
- [ ] After IPL complete call out both power supplies if either has TEMPERATURE_FAULT bit turn on (over temperature condition, paired supply may be putting out less current).
- [ ] After IPL complete, call out power supply if FAN_FAULT on (bad fan operation detected).
- [ ] After IPL complete, inform user redundancy is lost if any power supply presence indicates missing.

bjwyman, 2017-08-10:
Is there some overlap here with openbmc/openbmc#1860?

bjwyman, 2017-08-10:
Separate story? -> If power on fails due to time out or fault status, check STATUS_WORD for PG#, IOUT_OC_FAULT, VOUT_OV_FAULT, TEMPERATURE_FAULT, FAN_FAULT, or MFR_SPECIFIC. Log SEL pointing to the power supply that shows any of those fault bits.

bjwyman, 2017-08-11:
# **STATUS**
- 08/11:
  - Today:
  - Tomorrow:
  - Blockers:

bjwyman, 2017-08-31:
Removing this task "Update PMBus hwmon directory when power supply goes from missing to present."
It will be handled in openbmc/openbmc#2205.

bjwyman, 2017-09-08:
The desired both/all power supply call out for the temperature fault was noted as not an easy task. I communicated that fact with Jordan, and the plan to only call out the one supply reporting the fault. He has made a note about that.

rfrandse, 2017-09-26:
<https://gerrit.openbmc-project.xyz/6866> Change order of powered on fault checks
Resolves:  #1732 Monitor for faults at runtime

closed 2017-10-03 by geissonator