openbmc/openbmc #1737: Send SYNC commands to power supplies on a power on

spinler, 2017-06-06:


spinler, 2017-07-28:
This is for syncing up trending data between the 2 devices, so moving to phase 6 along with that function.

bjwyman, 2017-09-21:
This is in the #2310 Squad 2 list under my name. Was I supposed to have it assigned to me?

bjwyman, 2017-09-21:
Tried a manual test on a system in Rochester.
Odd results, packets seem to update faster than 30 seconds, among other odd results.

bjwyman, 2017-09-22:
# Task List
- [ ] Add service file to toggle "SYNC" line for power supplies at/around power on.
- [ ] Update recipe file(s) to include new service file.

rfrandse, 2018-01-19:
<https://gerrit.openbmc-project.xyz/6996> Create recipe to add service for power supply SYNC
Resolves:  #1737 Send SYNC commands to power supplies on a power on

closed 2018-01-31 by geissonator