openbmc/openbmc #1738: Add Fan Fault LEDs to device tree

spinler, 2017-06-06:
They're off of an IO expander

bjwyman, 2017-06-26:
https://lists.ozlabs.org/pipermail/openbmc/2017-June/008012.html

bjwyman, 2017-07-06:
v2 : https://lists.ozlabs.org/pipermail/openbmc/2017-July/008158.html

bjwyman, 2017-07-13:
Changes merged now.
https://lists.ozlabs.org/pipermail/openbmc/2017-July/008246.html
https://github.com/openbmc/linux/commit/31e55657b6faf93d8fb8b73763ec54446d3616d1

bjwyman, 2017-07-13:
Still some questions:
* Why is the pca955x: tacked onto the front of all the LED names?
* It was not obvious if the device driver supported hardware blink.
@legoater

mdmillerii, 2017-07-13:
@bjwyman of_read_string returns 0 on success and -ERROR on failure, so the test needs to be inverted
when [ reading the label from the device tree.](https://github.com/openbmc/linux/blob/c7e9e91a9be1ac4333051972dbc65e7cd2cea0dc/drivers/leds/leds-pca955x.c#L358) 

@legoater @shenki  this line first appears in [leds: pca955x: add device tree support](https://github.com/openbmc/linux/commit/73c3ca39b161eb901967af9b19c47b1ed05e418a).

mdmillerii, 2017-07-14:
Thinking about it again I retract my previous comment, if 0 the label is used.

bjwyman, 2017-07-15:
# STATUS
## July 14, 2017
* Not certain if this should be closed, or if we need something to address the blink and pca955x: label questions.

mdmillerii, 2017-07-17:
I did some research.  The 9555 driver for the chip uses one pwm for half brightness and the second for a custom brightness.  The leds-pca963x driver has a hack to enable hardware blink.  Its a hack because there are two pwms but it uses one to support the last led that requests a blink. 

Regardless, the user api is the same: set the trigger to blink then set the on / off ms to set a duty cycle and/or rate if desired (the default is 0 for both which is "driver chooses", most are doing 500ms on/ 500ms off).

The driver would need to pick up blink support and possibly the dts be enabled to opt into the non-compliant shared blink support.  It would reduce jitter.

mdmillerii, 2017-07-18:
Regarding the pca955x prefix:  After further study I found where the prefix was added and thought about what a patch would look like to retain user space compatibility with other systems.  Then I searched for more documentation and found [the documented naming convention](https://github.com/openbmc/linux/blob/dev-4.10/Documentation/leds/leds-class.txt#L41), which says our names are not compliant because they do not contain the device name and colour, just function.  (One might argue between the driver name and the device name).

The naming had convention was added [10 years ago](https://github.com/openbmc/linux/commit/6c152beefbf90579d21afc4f7e075b1f801f9a75).

bjwyman, 2017-07-19:
I had read that naming convention document after noticing the pca955x prefix. I was not sure if that counted as a "devicename" or not.
I did have the color in the label, yellow, but I removed that as an attempt to match up with the XML in the MRW, plus some argued that it should be amber.
@mdmillerii what do you think the labels should be? The way I read that, it seems like following the convention would get us to something like: pca955x:amber:fan0.

spinler, 2017-07-19:
Picked up by openbmc.  closing.

closed 2017-07-19 by spinler

bjwyman, 2017-07-19:
openbmc/openbmc#1987 opened to address the LED labeling issue.

bjwyman, 2017-07-19:
openbmc/openbmc#1988 opened to look into PCA955x device driver support of hardware blink.

bjwyman, 2017-07-19:
Linux bump that included the PCA955x addition for fan fault LEDs:
https://github.com/openbmc/openbmc/commit/8b60de789d7f553a3480db2590304fbc9a33d1c2

mdmillerii, 2017-07-19:
@bjwyman I agree `pca955x:amber:fan0`.