openbmc/openbmc #1739: Do Witherspoon-specific bitbake recipes in a non-Witherspoon-specific way

bjwyman, 2017-06-06:
Much discussion on best, more appropriate way to do some Witherspoon-specific changes in recipe file updates:

https://gerrit.openbmc-project.xyz/#/c/3991/29/meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-phosphor/fans/phosphor-fan%25.bbappend

@bradbishop

bjwyman, 2018-06-04:
Original link does not appear to work anymore. Gerrit server change? I think this is pointing to the same spot as the original comment:
https://gerrit.openbmc-project.xyz/#/c/3991/29/meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-phosphor/fans/phosphor-fan%2525.bbappend

gtmills, 2018-08-20:
@bradbishop @bjwyman  What is wrong with how we do System-specific bitbake recipes in Witherspoon?

bjwyman, 2019-03-29:
https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/3991/29/meta-openbmc-machines/meta-openpower/meta-ibm/meta-witherspoon/recipes-phosphor/fans/phosphor-fan%2525.bbappend#1

closed 2019-10-28 by stale[bot]