openbmc/openbmc #1740: phosphor-ipmi-host.service:  code dumped, status=6/ABRT [ Witherspoon]

gkeishin, 2017-06-07:
**Driver Info**

```
Build: https://openpower.xyz/job/openbmc-build/794/
VERSION_ID="v1.99.6-78-gfd4d3cc"
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Tue Jun 6 22:21:52 UTC 2017 armv6l GNU/Linux
```

Apr 20 17:41:29 witherspoon systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=6/ABRT
Apr 20 17:41:29 witherspoon systemd[1]: phosphor-ipmi-host.service: Unit entered failed state.
Apr 20 17:41:29 witherspoon systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'

```
● phosphor-ipmi-host.service - Phosphor Inband IPMI
   Loaded: loaded (/lib/systemd/system/phosphor-ipmi-host.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2017-04-20 17:41:30 UTC; 1min 1s ago
 Main PID: 1741 (ipmid)
   CGroup: /system.slice/phosphor-ipmi-host.service
           └─1741 /usr/sbin/ipmid
```
```
Apr 20 17:41:30 witherspoon systemd[1]: phosphor-ipmi-host.service: Service hold-off time over, scheduling restart.
Apr 20 17:41:30 witherspoon systemd[1]: Stopped Phosphor Inband IPMI.
Apr 20 17:41:30 witherspoon systemd[1]: Started Phosphor Inband IPMI.
Apr 20 17:41:31 witherspoon ipmid[1741]: Failed to call the method: Input/output error
Apr 20 17:41:31 witherspoon ipmid[1741]: Dest: :1.5, Path: /org/openbmc/HostIpmi/1
```

gkeishin, 2017-06-07:
- [x] Journald log 
[journald_v1.99.6-78-gfd4d3cc_ipmi_core.txt](https://github.com/openbmc/openbmc/files/1057031/journald_v1.99.6-78-gfd4d3cc_ipmi_core.txt)
-[x] Core dump tar zip file 
[core_dumps.tar.gz](https://github.com/openbmc/openbmc/files/1057036/core_dumps.tar.gz)

```
root@witherspoon:~# ls -lart /tmp/core_ipmid.1*
-rw-------    1 root     root        806912 Apr 20 17:37 /tmp/core_ipmid.1289
-rw-------    1 root     root        806912 Apr 20 17:38 /tmp/core_ipmid.1710
-rw-------    1 root     root        806912 Apr 20 17:38 /tmp/core_ipmid.1712
root@witherspoon:~#
```

gkeishin, 2017-06-07:
Test: Load the new driver and Poweron and the ipmid will crashed out..

dkodihal, 2017-06-07:
Yet to look at the core dumps, but see this in the journal :

Apr 20 17:37:29 witherspoon ipmid[1289]: Control Host work queue is empty!
Apr 20 17:37:29 witherspoon ipmid[1289]: terminate called after throwing an instance of 'phosphor::logging::xyz::openbmc_project::Control::Internal::Host::QueueEmpty'
Apr 20 17:37:29 witherspoon ipmid[1289]:   what():  xyz.openbmc_project.Control.Internal.Host.QueueEmpty

Any idea about this @geissonator?

geissonator, 2017-06-07:
Yeah, that's going to happen when the SMS attention gets queued up but the host doesn't see it.  The host then sees it on the next boot and we have an empty queue.  I need an issue for that one (clear the sms before each boot).  But, ipmid core dumped many other times during this boot.  I think we should use this bug to look into that.  This seems to be the first one:

```
Apr 20 17:32:58 witherspoon ipmid[920]: Successfully verified area checksum. offset:[8]
Apr 20 17:32:58 witherspoon ipmid[920]: SUCCESS: Populated FRU areas for:[/tmp/ipmifru1a]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Manufacturer] : Type Code = [BINARY] : Len = [0] : Val = []
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Name] : Type Code=[ASCII+Latin] : Len = [18] : Val = [OpenPOWER Firmware]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Model Number] : Type Code = [BINARY] : Len = [0] : Val = []
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Version] : Type Code=[ASCII+Latin] : Len = [34] : Val = [IBM-witherspoon-ibm-OP9_v1.16_1.24]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Serial Number] : Type Code = [BINARY] : Len = [0] : Val = []
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Asset Tag] : Type Code = [BINARY] : Len = [0] : Val = []
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [FRU File ID] : Type Code = [BINARY] : Len = [0] : Val = []
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Custom Field 1] : Type Code=[ASCII+Latin] : Len = [33] : Val = [        op-build-v1.16-83-ge67b53c-dirty]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Custom Field 2] : Type Code=[ASCII+Latin] : Len = [31] : Val = [        buildroot-2017.02.2-7-g23118ce]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Custom Field 3] : Type Code=[ASCII+Latin] : Len = [18] : Val = [        skiboot-5.6.0-rc2]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Custom Field 4] : Type Code=[ASCII+Latin] : Len = [17] : Val = [        hostboot-d02273f]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Custom Field 5] : Type Code=[ASCII+Latin] : Len = [33] : Val = [        linux-4.11.1-openpower1-pc11ec03]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Custom Field 6] : Type Code=[ASCII+Latin] : Len = [26] : Val = [        petitboot-v1.4.2-pcb6424c]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Custom Field 7] : Type Code=[ASCII+Latin] : Len = [19] : Val = [        machine-xml-e53653]
Apr 20 17:32:58 witherspoon ipmid[920]: _append_to_dict: VPD Key = [Custom Field 8] : Type Code = [BINARY] : Len = [0] : Val = []
Apr 20 17:32:58 witherspoon ipmid[920]: ERROR Unable to get the fru info for FRU=26
Apr 20 17:32:58 witherspoon ipmid[920]: Error updating inventory
Apr 20 17:32:59 witherspoon phosphor-gevent[984]: ::60a9:6c76:8e91:f402%1995942684 - - [2017-04-20 17:32:59] "POST /login HTTP/1.1" 200 320 0.035810
Apr 20 17:33:00 witherspoon phosphor-gevent[984]: ::60a9:6c76:8e91:f402%1995942684 - - [2017-04-20 17:33:00] "GET /xyz/openbmc_project/list HTTP/1.1" 200 11131 0.693909
Apr 20 17:33:00 witherspoon phosphor-gevent[984]: ::60a9:6c76:8e91:f402%1995942684 - - [2017-04-20 17:33:00] "POST /logout HTTP/1.1" 200 184 0.023618
Apr 20 17:33:11 witherspoon obmc-console-server[831]: obmc-console-server: Write error: Resource temporarily unavailable
Apr 20 17:33:11 witherspoon obmc-console-server[831]: obmc-console-server: Write error: Resource temporarily unavailable
Apr 20 17:33:11 witherspoon obmc-console-server[831]: obmc-console-server: Write error: Resource temporarily unavailable
Apr 20 17:33:11 witherspoon obmc-console-server[831]: obmc-console-server: Write error: Resource temporarily unavailable
Apr 20 17:33:11 witherspoon systemd[1]: phosphor-ipmi-host.service: Main process exited, code=dumped, status=11/SEGV
Apr 20 17:33:12 witherspoon systemd[1]: phosphor-ipmi-host.service: Unit entered failed state.
Apr 20 17:33:12 witherspoon systemd[1]: phosphor-ipmi-host.service: Failed with result 'core-dump'.
Apr 20 17:33:12 witherspoon systemd[1]: phosphor-ipmi-host.service: Service hold-off time over, scheduling restart.
Apr 20 17:33:12 witherspoon systemd[1]: Stopped Phosphor Inband IPMI.
```

@nasamuffin - Anything jump out at you here?  You and I both put some new code into phosphor-host-ipmid yesterday it looks like.

geissonator, 2017-06-12:
```
Apr 22 09:22:21 witherspoon ipmid[958]: Failed to get /system/chassis/motherboard/cpu0 busname:
Apr 22 09:22:21 witherspoon ipmid[958]: *** Error in `/usr/sbin/ipmid': free(): invalid pointer: 0x424c5b2c ***
```

closed 2017-06-13 by williamspatrick

geissonator, 2017-06-13:
Looks like Patrick identified a fix in https://gerrit.openbmc-project.xyz/#/c/4753/