openbmc/openbmc #1741: avahi-daemon: Service name conflict for "phosphor_rest"/sftp/obmc_console [ Witherspoon]

gkeishin, 2017-06-07:
On boot to BMC ready, we see this bunch of conflicts in the journald

```
Apr 20 17:30:13 witherspoon avahi-daemon[826]: Service name conflict for "phosphor_rest" (/services/phosphor_rest.service), retrying with "phosphor_rest #3".
Apr 20 17:30:13 witherspoon avahi-daemon[826]: Service name conflict for "sftp" (/services/sftp.service), retrying with "sftp #3".
Apr 20 17:30:13 witherspoon avahi-daemon[826]: Service name conflict for "obmc_console" (/services/obmc_console.service), retrying with "obmc_console #3".
Apr 20 17:30:14 witherspoon avahi-daemon[826]: Service name conflict for "phosphor_rest" (/services/phosphor_rest.service), retrying with "phosphor_rest #4".
Apr 20 17:30:14 witherspoon avahi-daemon[826]: Service name conflict for "sftp" (/services/sftp.service), retrying with "sftp #4".
Apr 20 17:30:14 witherspoon avahi-daemon[826]: Service name conflict for "obmc_console" (/services/obmc_console.service), retrying with "obmc_console #4".
```

**Driver Info**

```
Build: https://openpower.xyz/job/openbmc-build/794/
VERSION_ID="v1.99.6-78-gfd4d3cc"
Linux witherspoon 4.10.5-8736136e756ec127de0bbe2c7e2de683204d4512 #1 Tue Jun 6 22:21:52 UTC 2017 armv6l GNU/Linux
```

- [x]  Journald 
[journald_v1.99.6-78-gfd4d3cc_avahi.txt](https://github.com/openbmc/openbmc/files/1057078/journald_v1.99.6-78-gfd4d3cc_avahi.txt)

geissonator, 2017-11-09:
This takes up to 20% of our journal on an boot, be really good to see if there's a config option or something we can compile avahi with to reduce it's logging.

gkeishin, 2018-04-24:
@ratagupt do we still need this one.. I dont see in recent builds .

closed 2019-10-28 by stale[bot]