openbmc/openbmc #1742: Seeing SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590) on witherspoon

PavamanSubramaniyam, 2017-06-07:
I have tried to code update the latest OpenBMC Firmware image on witherspoon P9 Hardware.
Seeing the below messages in OpenBMC shell login.

```
Jun 07 08:09:14 witherspoon phosphor-bmc-state-manager[972]: BMC_READY
Jun 07 08:09:14 witherspoon phosphor-bmc-state-manager[972]: Setting the BMCState field
Jun 07 08:09:17 witherspoon phosphor-gevent[908]: Traceback (most recent call last):
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:     result = self._run(*self.args, **self.kwargs)
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:     ciphers=ciphers)
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:     self.do_handshake()
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Jun 07 08:09:17 witherspoon phosphor-gevent[908]:     return self._sslobj.do_handshake()
Jun 07 08:09:17 witherspoon phosphor-gevent[908]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Jun 07 08:09:17 witherspoon phosphor-gevent[908]: <Greenlet at 0x76086300: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x760608b0 fileno=5 address=('::3432:f776:b00d:6d76%1986858400', 4
43, 0, 1986858400)>>(<socket at 0x75f352d0 fileno=9>, ('::6019:6b76:8e91:f402%1995840284', 33622, 0, 199)> failed with SSLEOFError
```

PavamanSubramaniyam, 2017-06-07:
The machine is installed with following levels of Open BMC and PNOR Firmware images.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-78"
VERSION_ID="v1.99.6-78-gfd4d3cc"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-78"
BUILD_ID="v1.99.6"

root@witherspoon:~# pflash -r /dev/stdout -P VERSION
Reading to "/dev/stdout" from 0x027e0000..0x027e1000 !
[                                                  ] 0%open-power-witherspoon-v1.17-25-g4711f01
        buildroot-2017.02.2-7-g23118ce
        skiboot-5.6.0-16-g9abb449b9c57
        hostboot-d96098e
        linux-4.11.2-openpower1-p8f7ea46
        petitboot-v1.4.2-pb361a08
        machine-xml-96746e9
        occ-bc34ad5
        hostboot-binaries-7629121
        capp-ucode-9c73e9f
        sbe-d770027
[==================================================] 100%
```

PavamanSubramaniyam, 2017-06-07:
[journalctl.txt](https://github.com/openbmc/openbmc/files/1057388/journalctl.txt)

gkeishin, 2017-06-07:
Isn't this the same with https://github.com/openbmc/openbmc/issues/1367

geissonator, 2017-06-07:
Maybe the same?  I think in the past we've blamed this on the ssh client.  @legoater is this something your change in #1367 will address?

PavamanSubramaniyam, 2018-08-28:
I have a P9 Witherspoon DD2.1 Hardware with latest CORAL Firmware driver installed on the machine.

```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="ibm-v2.0"
VERSION_ID="ibm-v2.0-0-r46-0-gbed584c"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) ibm-v2.0"
BUILD_ID="ibm-v2.0-0-r46"

root@witherspoon:~# cat /var/lib/phosphor-software-manager/pnor/ro/VERSION
IBM-witherspoon-ibm-OP9_v1.19_1.192
        op-build-v2.0.5-343-g6f85520
        buildroot-2018.02.1-6-ga8d1126
        skiboot-v6.0.8
        hostboot-4bd54cb-p8235b67
        occ-f796766
        linux-4.16.13-openpower1-p4213cac
        petitboot-v1.7.2-p439439e
        machine-xml-94a137f
        hostboot-binaries-d01b024
        capp-ucode-p9-dd2-v4
        sbe-defe254

```

Still seeing the same issue in the journalctl logs on the machine.

```
Aug 28 05:05:14 witherspoon phosphor-host-state-manager[1418]: External request to reset reboot count
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]: Traceback (most recent call last):
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:   File "/usr/lib/python2.7/site-packages/gevent/greenlet.py", line 327, in run
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:     result = self._run(*self.args, **self.kwargs)
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:   File "/usr/lib/python2.7/site-packages/gevent/server.py", line 102, in wrap_socket_and_handle
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:     ssl_socket = self.wrap_socket(client_socket, **self.ssl_args)
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 404, in wrap_socket
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:     ciphers=ciphers)
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 113, in __init__
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:     self.do_handshake()
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:   File "/usr/lib/python2.7/site-packages/gevent/ssl.py", line 324, in do_handshake
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]:     return self._sslobj.do_handshake()
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]: SSLEOFError: EOF occurred in violation of protocol (_ssl.c:590)
Aug 28 05:05:14 witherspoon phosphor-gevent[1256]: <Greenlet at 0x756a8030: <bound method WSGIServer.wrap_socket_and_handle of <WSGIServer at 0x760d6eb0 file
no=6 address=('::34c2:744:ae:b976%1991880176', 443, 0, 1991880176)>>(<socket at 0x765560d0 fileno=10>, ('::c0a9:b776:8e91:f402%1141287708', 52790, 0, 114)> f
ailed with SSLEOFError
Aug 28 05:05:14 witherspoon phosphor-fan-control[12741]: [[0;1;31mUnable to find property on interface[[0m
Aug 28 05:05:16 witherspoon phosphor-fan-control[12741]: [[0;1;31mUnable to find property on interface[[0m
Aug 28 05:05:18 witherspoon phosphor-fan-control[12741]: [[0;1;31mUnable to find property on interface[[0m
```

closed 2019-04-26 by stale[bot]