openbmc/openbmc #1743: Intermittent failure finding soft power off dbus interface

geissonator, 2017-06-07:
Saw this pop up on a system this morning, where the AES release job was just trying to power a system down.

```
Jun 07 03:41:37 witherspoon systemd[1]: Starting Soft power off of the host...
Jun 07 03:41:37 witherspoon phosphor-ledmanager[871]: No service providers for physical LED
Jun 07 03:41:37 witherspoon phosphor-ledmanager[871]: De-Asserting LEDs
Jun 07 03:41:37 witherspoon phosphor-ledmanager[871]: No service providers for physical LED
Jun 07 03:41:37 witherspoon systemd[1]: Started Assert power_on LED.
Jun 07 03:41:37 witherspoon ipmid[1360]: Pushing cmd on to queue
Jun 07 03:41:37 witherspoon ipmid[1360]: Asserting SMS Attention
Jun 07 03:41:37 witherspoon ipmid[1360]: SMS Attention asserted
Jun 07 03:41:37 witherspoon ipmid[1360]: Failed to get /xyz/openbmc_project/ipmi/internal/soft_power_off bus name: No such file or directory
Jun 07 03:41:37 witherspoon ipmid[1360]: Did not find soft off service so request Host:Transition:Off
Jun 07 03:41:37 witherspoon phosphor-softpoweroff[1424]: Host control signal values
Jun 07 03:41:37 witherspoon phosphor-softpoweroff[1424]: Timer started waiting for host to shutdown
Jun 07 03:41:37 witherspoon phosphor-host-state-manager[979]: Host State transaction request
Jun 07 03:41:37 witherspoon ipmid[1360]: Transition request initiated successfully
Jun 07 03:45:47 witherspoon phosphor-host-state-manager[979]: Host State transaction request
Jun 07 03:45:54 witherspoon systemd[1]: Starting Cleanup of Temporary Directories...
Jun 07 03:45:54 witherspoon systemd[1]: Started Cleanup of Temporary Directories.
Jun 07 04:26:37 witherspoon phosphor-softpoweroff[1424]: Timer expired
Jun 07 04:26:37 witherspoon phosphor-softpoweroff[1424]: Host did not shutdown within configured time.
Jun 07 04:26:38 witherspoon systemd[1]: xyz.openbmc_project.Ipmi.Internal.SoftPowerOff.service: Main process exited, code=exited, status=255/n/a
Jun 07 04:26:38 witherspoon systemd[1]: Failed to start Soft power off of the host.
Jun 07 04:26:38 witherspoon systemd[1]: Dependency failed for Power0 Host Off.
```

The soft power off service was clearly running because it was what caused the SMS attention.  Not sure if maybe there's some sort of delay required to ensure mapper has registered the object first?  I assume that's the issue here, soft power off is on the dbus but mapper just hasn't registered it yet, and so when ipmid asks mapper, it says can't find it.

geissonator, 2017-06-07:
Verified it shows up when I look after the fail so a race condition seems like the issue here:
```
[witherspoon]# busctl call xyz.openbmc_project.ObjectMapper /xyz/openbmc_project/object_mapper xyz.openbmc_project.ObjectMapper GetObject sas "/xyz/openbmc_project/ipmi/internal/soft_power_off" 1 "xyz.openbmc_project.Ipmi.Internal.SoftPowerOff"
a{sas} 1 ":1.125" 1 "xyz.openbmc_project.Ipmi.Internal.SoftPowerOff"
```

geissonator, 2017-06-07:
Potential fix here - https://gerrit.openbmc-project.xyz/#/c/4649/

rfrandse, 2017-06-08:
<https://gerrit.openbmc-project.xyz/4649> Ensure softoff in mapper
Resolves  #1743 Intermittent failure finding soft power off dbus interface

geissonator, 2017-06-08:
We discussed my proposed solution this morning.  The general agreement among the team was that the refactor of mapper in #1661 will provide the function I require for this bug.  Basically, we need mapper to guarantee it process's signals and interface requests in order.  There was still some concern with situations where the new interface signal does not make it to mapper before a request for the interface makes it but the team didn't feel it was insurmountable.

So for this bug, I will simply be directly calling the dbus interface instead of using mapper and referencing the mapper refactor as a TODO to change back once available.

closed 2017-06-08 by williamspatrick