openbmc/openbmc #1745: [similar to Barreleye] The OpenBMC broken starting from June 2017

yadron, 2017-06-07:
If the version of OpenBmc of ~ 25 May worked well (that was 99.6.2), the latest (starting from ~2 June) fails with the same error "cannot mount /dev/mtdblock"; what is this and how to fix it? Thank you.

Log (and attached more detailed):

```
rtc-ds1307 11-0068: setting system clock to 2017-06-07 16:39:51 UTC (1496853591)
Freeing unused kernel memory: 1024K
grep: /sys/class/mtd/*/name: No such file or directory
cp: omitting directory '/run/mtd:u-boot-env'
grep: /sys/class/mtd/*/name: No such file or directory
grep: /sys/class/mtd/*/name: No such file or directory
cat: can't open '/run/fw_env': No such file or directory
cat: can't open '/run/fw_env': No such file or directory
rofs = squashfs rwfs = jffs2
mount: mounting /dev/mtdblock on run/initramfs/ro failed: No such file or directory
mount: mounting /dev/mtdblock on run/initramfs/rw failed: No such file or directory
random: fast init done

Mounting read-write /dev/mtdblock filesystem failed.  Please fix and run
        mount /dev/mtdblock run/initramfs/rw -t jffs2 -o rw
to to continue, or do change nothing to run from RAM for this boot.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):
```
[june_barreleye_nomtd_broken.txt](https://github.com/openbmc/openbmc/files/1058725/june_barreleye_nomtd_broken.txt)

geissonator, 2017-06-07:
If we hit something like this we'd probably pull the flash chip and manually reflash it.  Not sure if that's an option for you.  We know latest code is working ok on our systems, so something must have gone wrong in the code update.  Thoughts @shenki or @amboar ?

yadron, 2017-06-08:
I think that this change broke our platform:

`diff may25/openbmc/meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig ~june7/openbmc/meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig`:

```
...
< CONFIG_ASPEED_LPC_MBOX=y
---
> CONFIG_ASPEED_MBOX=y
...
```

or maybe this too:

```
73c73
< CONFIG_SPI_ASPEED_SMC=y
---
> CONFIG_ASPEED_FLASH_SPI=y
...
```

I'll check this and let you know.

yadron, 2017-06-08:
It looks like not "CONFIG_SPI_ASPEED_SMC" param problem, but other. Which one of these:

```
106a107
> CONFIG_REALTEK_PHY=y
127c128,129
< # CONFIG_HW_RANDOM is not set
---
> CONFIG_HW_RANDOM=y
> CONFIG_HW_RANDOM_TIMERIOMEM=y
129a132,134
> CONFIG_I2C_MUX=y
> CONFIG_I2C_MUX_PCA9541=y
> CONFIG_I2C_MUX_PCA954x=y
130a136,143
> CONFIG_I2C_FSI=y
> CONFIG_GPIOLIB=y
> CONFIG_GPIO_SYSFS=y
> CONFIG_GPIO_ASPEED=y
> CONFIG_W1=y
> CONFIG_W1_MASTER_GPIO=y
> CONFIG_W1_SLAVE_THERM=y
> CONFIG_SENSORS_IIO_HWMON=y
133d145
< CONFIG_SENSORS_POWER8_OCC_I2C=y
136a149,150
> CONFIG_SENSORS_LM25066=y
> CONFIG_SENSORS_UCD9000=y
137a152
> CONFIG_SENSORS_ASPEED=y
141a157
> CONFIG_LEDS_CLASS_FLASH=y
148a165,166
> CONFIG_RTC_DRV_PCF8523=y
> CONFIG_RTC_DRV_RV8803=y
151c169
< CONFIG_ASPEED_MBOX=y
---
> CONFIG_ASPEED_LPC_MBOX=y
152a171,173
> CONFIG_IIO=y
> CONFIG_ASPEED_ADC=y
> CONFIG_BMP280=y
154a176
> CONFIG_FSI_MASTER_HUB=y
156d177
< CONFIG_FSI_I2C=y
```

?

amboar, 2017-06-09:
> If we hit something like this we'd probably pull the flash chip and manually reflash it.

That's generally unnecessary unless you've bricked u-boot. If your kernel configuration is busted, you can netboot a new kernel from u-boot, testing until you get it right. This is possible on hardware and under qemu with the palmetto machine model if Barreleye (AST2400) is your concern.

> diff may25/openbmc/meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig ~june7/openbmc/meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig

@yadron This feels a little concerning from a workflow perspective - I'm surprised you're not using `git` to manage your state. Related, I find the output of `diff -u` easier to parse.

The defconfig changes you listed are all for the 4.10 kernel. Is this the kernel you are building? You haven't mentioned the version anywhere. As you note a number of the Kconfig symbols have slightly different names - this is a result of the patches being reviewed and applied upstream. Are you trying to build from openbmc master? You haven't mentioned at what commit you're trying to the build. Do you have patches on top?

What does your devicetree look like? It may be the case that you are missing changes there. Can you link me to a copy?

yadron, 2017-06-09:
@amboar , thank you for the comments.

Let me explain what I do. I have compiled OpenBMC "v1.99.6-2" (fcaf72e or "v1.99.6-2-gfcaf72e") on ~ 25 May 2017. It works fine. No problems with flash (mtd block). The host boots well.

I've tried compiling next master version on 2 June 2017 (the same way, see https://openpowerblog.wordpress.com/2016/04/23/how-to-build-openbmc-binary-for-barreleye/ ). It fails. Cannot boot BMC with cannot mount /dev/mtdblock error.

If I revert back to image-bmc from 25 May, then it works fine again. So no problems with flash itself. The only problem that I have to use external programmer for writing flash with older 25 May image-bmc instead of June 2 or June 7 OpenBMC, because if cannot boot OpenBMC, then cannot run ./update (so have to use external programmator for writing flash with older image-bmc then).

As result, the task is simple: I can compare what is the difference between 25 May version and 2+ June version. Then I can revert it one by one for finding which diff broken OpenBMC for Barreleye.

Versions of Kernel and most of software are the same, because not so much diff between 25 May master version and June master version (just few days passed and there were no big diff).

The only diff I see between 25 May and 7 June:

defconfig:

```
./meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig
--- /home/build/openbmc-msi5/openbmc/./meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig      2017-0
5-25 15:30:16.330000000 +0300
+++ ./meta-openbmc-bsp/meta-aspeed/meta-ast2400/recipes-kernel/linux/linux-obmc/defconfig       2017-06-09 11:23:10.579000000 +0300
@@ -70,7 +70,7 @@
 CONFIG_MTD_BLOCK=y
 CONFIG_MTD_PARTITIONED_MASTER=y
 CONFIG_MTD_SPI_NOR=y
-CONFIG_ASPEED_FLASH_SPI=y
+CONFIG_SPI_ASPEED_SMC=y
 CONFIG_MTD_UBI=y
 CONFIG_MTD_UBI_FASTMAP=y
 CONFIG_MTD_UBI_BLOCK=y
@@ -104,6 +104,7 @@
 # CONFIG_NET_VENDOR_VIA is not set
 # CONFIG_NET_VENDOR_WIZNET is not set
 CONFIG_BROADCOM_PHY=y
+CONFIG_REALTEK_PHY=y
 # CONFIG_WLAN is not set
 # CONFIG_INPUT_LEDS is not set
 # CONFIG_INPUT_MOUSEDEV is not set
@@ -124,21 +125,36 @@
 CONFIG_SERIAL_OF_PLATFORM=y
 CONFIG_SERIAL_ASPEED_VUART=y
 CONFIG_ASPEED_BT_IPMI_BMC=y
-# CONFIG_HW_RANDOM is not set
+CONFIG_HW_RANDOM=y
+CONFIG_HW_RANDOM_TIMERIOMEM=y
 CONFIG_I2C=y
 CONFIG_I2C_CHARDEV=y
+CONFIG_I2C_MUX=y
+CONFIG_I2C_MUX_PCA9541=y
+CONFIG_I2C_MUX_PCA954x=y
 CONFIG_I2C_ASPEED=y
+CONFIG_I2C_FSI=y
+CONFIG_GPIOLIB=y
+CONFIG_GPIO_SYSFS=y
+CONFIG_GPIO_ASPEED=y
+CONFIG_W1=y
+CONFIG_W1_MASTER_GPIO=y
+CONFIG_W1_SLAVE_THERM=y
+CONFIG_SENSORS_IIO_HWMON=y
 CONFIG_SENSORS_LM75=y
 CONFIG_SENSORS_NCT7904=y
-CONFIG_SENSORS_POWER8_OCC_I2C=y
 CONFIG_PMBUS=y
 # CONFIG_SENSORS_PMBUS is not set
 CONFIG_SENSORS_ADM1275=y
+CONFIG_SENSORS_LM25066=y
+CONFIG_SENSORS_UCD9000=y
 CONFIG_SENSORS_TMP421=y
+CONFIG_SENSORS_ASPEED=y
 # CONFIG_HID is not set
[... 174 lines cut ...]
+# change in polarity for the value matching user input. Once the required
+# condition is met, it calls into a systemd target file if one is mentioned
+# in input.
+PREFERRED_PROVIDER_virtual/obmc-gpio-monitor ?= "phosphor-gpio-monitor"
+
 OBMC_DBUS_PATH_ROOT ?= "/xyz/openbmc_project"
 OBMC_DBUS_IFACE_ROOT ?= "xyz.openbmc_project"
./meta-phosphor/conf/distro/openbmc-phosphor.conf
--- /home/build/openbmc-msi5/openbmc/./meta-phosphor/conf/distro/openbmc-phosphor.conf  2017-05-25 15:30:16.394000000 +0300
+++ ./meta-phosphor/conf/distro/openbmc-phosphor.conf   2017-06-09 11:23:10.662000000 +0300
@@ -23,7 +23,6 @@
         obmc-phosphor-fan-mgmt \
         obmc-phosphor-chassis-mgmt \
         obmc-phosphor-flash-mgmt \
-        obmc-phosphor-event-mgmt \
         obmc-phosphor-user-mgmt \
         obmc-settings-mgmt \
         obmc-phosphor-system-mgmt \
./meta-phosphor/conf/layer.conf
./meta-phosphor/conf/local.conf.sample
./meta-phosphor/conf/machine/include/obmc-bsp-common.inc
./openbmc-env
```

I'll let you know when I find which one of these broken Barreleye (the current problem is that dev/mtdblock missed, so for sure this is related to the Linux kernel of BMC).

yadron, 2017-06-09:
At last, I see that there was kernel update and the reason probably is inside kernel itself, but not in original OpenBMC configuration.

It was:

-rw-r--r--. 1 build build 8100 Jun  2 12:27 /home/build/openbmc-5/openbmc/build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.7.10+git999-r0/linux-obmc-4.7.10+git999/drivers/mtd/mtdblock.o

Now:

ls -l work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.10.5+git999-r0/linux-obmc-4.10.5+git999/drivers/mtd/mtdblock.o -rw-r--r--. 1 build build 8500 Jun  9 13:30 work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.10.5+git999-r0/linux-obmc-4.10.5+git999/drivers/mtd/mtdblock.o

Most likely that there are new params in Kernel and defconfig needs additional MTD params that were not required before.

But the only diff regarding MTD  is this param: "CONFIG_OF_MTD=y"
(that is the result of
```
[build@obmc openbmc]$ diff ../../openbmc-5/openbmc/build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.7.10+git999-r0/linux-obmc-4.7.10+git999/.config ./build/tmp/work/barreleye-openbmc-linux-gnueabi/linux-obmc/4.10.5+git999-r0/linux-obmc-4.10.5+git999/.config|grep MTD
< CONFIG_OF_MTD=y
```
)

amboar, 2017-06-09:
@yadron okay, do you mind sending a kernel defconfig patch to the openbmc list?

yadron, 2017-06-09:
I'm not sure if this is only param missed. Possibly something was removed from kernel.
Also, the diff is that:

< #define CONFIG_ASPEED_FLASH_SPI 1

< #define CONFIG_ASPEED_MBOX 1

- so in kernel 4.10 these params missed. Need to check if kernel 4.10 still supports ASPEED_FLASH that is probably required for barreleye.

sdemirhan, 2017-06-15:
Do you find the problem? I also faced the same problem with you and i did not boot openbmc. So if there is any progress please let me know.

mdmillerii, 2017-06-15:
Sorry I started a post from my phone but got sidetracked.  This says "Similar to Barreleye" so I assume that this is some other ast2400 system for now.  My understanding is that Barreleye is working with the additional patches to linux in the openbmc repository.

The fact that the grep for /sys/class/mtd/*/name is not matching any files says your spi controller drivers are not binding or not identifying any attached nor chips.

The upstream `CONFIG_SPI_ASPEED_SMC` driver replaces the `CONFIG_ASPEED_FLASH_SPI` development version.  Similarly `CONFIG_ASPEED_MBOX` became `CONFIG_ASPEED_LPC_MBOX`.

In addition to the patch that fixes the device tree include factoring  `file://v2-0001-arm-dts-aspeed-Add-missing-clock-sources-for-barr.patch` you need to make sure that `CONFIG_VMSPLIT_2G` is set to allow enough kernel address space to map the entire 256MB window of each controller (vs the 64MB or 32MB of flash chips that were previously mapped with the old controller) as the driver now manages the mapping of chips into the window.

Also be sure to remove any reg resources that may be causing limited resoruces to be available to the driver.   You may want to look for an error number from the driver trying to initialize.

yadron, 2017-06-16:
@mdmillerii , thank you for the comments.

I've just got the patch file://v2-0001-arm-dts-aspeed-Add-missing-clock-sources-for-barr.patch and successfully applied it to latest OpenBMC/Barreleye (the resulted version is "v1.99.7-5-g1afb614-dirty").

I'll let you know as soon as I check it with Barreleye.

yadron, 2017-06-19:
I've tried latest OpenBMC (v1.99.7-5) with patch "0002-arm-dts-aspeed-Add-missing-clock-sources-for-barr.patch", and I can boot BMC at Barreleye (and I can boot host).

As result, this ticket can be closed as soon as patch "v2-0001-arm-dts-aspeed-Add-missing-clock-sources-for-barr.patch" would be included to OpenBMC.

Currently I had to do next workaround for applying this patch:

"successfully built Barreleye image with this patch being applied (FYI, I've copied patch to "import-layers/yocto-poky/meta/recipes-kernel/sysprof/files/0002-arm-dts-aspeed-Add-missing-clock-sources-for-barr.patch" and then added this filename to SRC_URI of import-layers/yocto-poky/meta/recipes-kernel/sysprof/sysprof_git.bb and then recompiled Barreleye image). The image version is "v1.99.7-5-g1afb614-dirty".

mdmillerii, 2017-06-19:
The patch is in the dev-4.10 branch and was merged into openbmc by 169657e29c16309a5fc1e66f3ebee720930f8663.

closed 2017-06-19 by williamspatrick