openbmc/openbmc #1746: org.openbmc.settings.Host.service keep restarting on AST 2500 eval board

mohitguptams, 2017-06-07:
I am seeing below error continuously in journalctl and setting manager is taking 100% of CPU. 

Any ideas what is going wrong? I did not make any changes.

*********
2223     1 root     R    13216   3% 100% {settings_manage} /usr/bin/python -u 

**** Journalctl log
May 25 12:08:23 evb-ast2500 settings_manager.py[20308]:   File "/usr/sbin/settings_manager.py", line 11, in <module>
May 25 12:08:23 evb-ast2500 settings_manager.py[20308]:     from IPy import IP
May 25 12:08:23 evb-ast2500 settings_manager.py[20308]: ImportError: No module named IPy
May 25 12:08:23 evb-ast2500 systemd[1]: [[0;1;39morg.openbmc.settings.Host.service: Main process exited, code=exited, status=1/FAILURE[[0m
May 25 12:08:23 evb-ast2500 systemd[1]: [[0;1;31mFailed to start Phosphor Settings Manager.[[0m
May 25 12:08:23 evb-ast2500 systemd[1]: [[0;1;39morg.openbmc.settings.Host.service: Unit entered failed state.[[0m
May 25 12:08:23 evb-ast2500 systemd[1]: [[0;1;39morg.openbmc.settings.Host.service: Failed with result 'exit-code'.[[0m
May 25 12:08:23 evb-ast2500 systemd[1]: org.openbmc.settings.Host.service: Service hold-off time over, scheduling restart.
May 25 12:08:23 evb-ast2500 systemd[1]: Stopped Phosphor Settings Manager.
May 25 12:08:23 evb-ast2500 systemd[1]: Starting Phosphor Settings Manager...
May 25 12:08:25 evb-ast2500 settings_manager.py[20311]: Traceback (most recent call last):
May 25 12:08:25 evb-ast2500 settings_manager.py[20311]:   File "/usr/sbin/settings_manager.py", line 11, in <module>

geissonator, 2017-06-07:
Did you somehow delete /usr/lib/python2.7/site-packages/IPy.py  ?  That's where it's located on the BMC filesystem.

mohitguptams, 2017-06-07:
I did not delete it. But I also did not have it on BMC System.

[image]

But i do see IPy.py in my build at following location

build/tmp/sysroots/evb-ast2500/usr/lib/python2.7/site-packages

And i also see that there are others py file like inflection.py which are in my build but not on the system. Is there more strip down version of site-packages somewhere in the build? and if yes is there any configuration controlling that?

mdmillerii, 2017-06-15:
You should check your overlay that it was not deleted there.  You can look either under the covers at `/run/initramfs/rw/cow/<path>` or you can wipe the files in the overlay.  (You should find the base file at `/run/initramfs/ro/<path>` where `<path>` is `/usr/lib/python2.7/site-packages/IPy.py`.

gtmills, 2018-05-23:
@mohitguptams @mdmillerii Still a problem? Can this be closed?

mohitguptams, 2018-05-23:
Yes please we can close this.

closed 2018-05-23 by gtmills