openbmc/openbmc #1747: Master Zaius build doesn't pick up the Mac address from VPD

agangidi53, 2017-06-07:
I built the latest openBMC build form master for zaius and it doesn't seem to pick the mac address form VPD . Instead generates random mac address. 

I followed the usual steps to build that I usually do:

root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="v1.99.6-78"
VERSION_ID="v1.99.6-78-gfd4d3cc-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) v1.99.6-78"
BUILD_ID="v1.99.6"

**Journal Log Snippet:**

'''
Jun 07 17:14:24 zaius kernel: ftgmac100 1e660000.ethernet: Generated random MAC address 16:ca:66:fb:42:82
Jun 07 17:14:24 zaius kernel: ftgmac100 1e660000.ethernet: Using NCSI interface
Jun 07 17:14:24 zaius kernel: ftgmac100 1e660000.ethernet eth0: irq 20, mapped at f0060000
Jun 07 17:14:24 zaius kernel: ftgmac100 1e680000.ethernet: Generated random MAC address c6:7e:97:58:7c:bb
Jun 07 17:14:24 zaius kernel: libphy: ftgmac100_mdio: probed
Jun 07 17:14:24 zaius kernel: Broadcom BCM54612E 1e680000.etherne:01: attached PHY driver [Broadcom BCM54612E] (mii_bus:phy_addr=1e
680000.etherne:01, irq=-1)
Jun 07 17:14:24 zaius kernel: ftgmac100 1e680000.ethernet eth1: irq 21, mapped at f0080000
Jun 07 17:14:24 zaius kernel: .[0;1;39m fsi0: can't read slave 00:00 -5.[0m
Jun 07 17:14:24 zaius kernel: i2c /dev entries driver
'''

geissonator, 2017-06-07:
I believe this is what @chinaridinesh is working on in #1315 ?

agangidi53, 2017-06-09:
@geissonator Could be. But the timeline of the comments there is a bit messy. I had a openBMC build from April 17th have this work okay and a new build I built yesterday had this issue.

I'll dig deeper. Thanks

agangidi53, 2017-07-11:
Any update on this ?

agangidi53, 2018-05-31:
This is fixed thanks. Can be closed.

closed 2018-05-31 by gtmills