openbmc/openbmc #1748: Sprint 17.06.14 - Committed

rfrandse, 2017-06-07:


dkodihal, 2017-06-09:
Squad 1 tentative sprint plan

Ratan
#1678 Settings/Network interface redesign
Implementation story for #1685 Networkd:-Define the DBUS interface for DHCP option configuration

Tom
#1515  SEL: Implement the mandatory SEL commands wil move
Find another one for the actual sprint

Devender
#1560 IPMI FRU commands : implementation will move

Vishwa
#1363 Request OCC reset when fail detected 

Jayanth
#1508 REST-API: Add File read command
Another dump one in actual?

Dhruv

Deepak

bradbishop, 2017-06-13:
-------------------------------------------------------------------------------------------------
***Squad 2 -***
-------------------------------------------------------------------------------------------------
-------------------------------------------------------------------------------------------------
**Gunnar:gtmills** 
**17.05.31:**
- **#1610 (legacy fans)**

-------------------------------------------------------------------------------------------------
**Matt S:spinler** 
**17.05.31:**
- #1342 (base Watchdog support - install and run busybox watchdog) 
- **#1687 (pulse watchdog reset on power-on  (requires phosphor-gpio-util))**
- **#1688  (create OnFailure unit monitor to stop watchdog)**

-------------------------------------------------------------------------------------------------
**Matt B:msbarth**
**17.05.31:**
- #1626 (calculate speed change request)
- #1625 (loop)
- **#940 (fan floor)**
- **#1628 (ceilings)**

-------------------------------------------------------------------------------------------------
**Brad:bradbishop**
**17.05.31:**
- #1318 (gpio presence)
- #1300 (real error logs in presence)

-------------------------------------------------------------------------------------------------
**Chris:cbostic**
**17.05.03:**
- #1599 (perf improvements)

**17.05.17:**
- #1177 (Add IR35221 to Witherspoon device tree)

**17.05.31:**
- #1635 (Non hwmon attrs for MAX)

- #1636 (watchdog driver sysfs)

- #1702 (unbind ir35221)

------------------------------------------------------------------------------------------------
**Eddie:eddiejames**

**17.05.31:**

- #1698 - fsi rescan for occ devices

- #1699 - second processor occ doesn't work

- #1619 (non hwmon OCC sysfs attributes)

-------------------------------------------------------------------------------------------------
**Brandon:bjwyman**

**17.05.30:**
 - #1738 (fan fault led device)
-------------------------------------------------------------------------------------------------
**Dinesh:chinaridinesh**
**17.03.08:**
- #1315 (Ensure uboot fw_env update script uses new inventory object) 

**17.05.03:**
- #1322 (pgood is lost on BMC reboot on witherspoon)

**17.05.31:**
- #1555 (make real error logs in fan control)

anoo1, 2017-06-13:
# Squad 3 -

**Adriana:anoo1 -**

**17.03.08**
- #1297 eSEL Missing AdditionalData [ Witherspoon ]
  * 06/12: Patch needs reword per Patrick.

**17.05.31**
- #1651 BMC filesystem: Create UBI read-only and read-write partitions
  * 6/05: Got details from Patrick.

**17.06.14**
- #1673 Refactor ubi service files into shell scripts
- #1758 BMC: Flag to specify the chips where partitions should be stored

---

**Andrew:geissonator -**

**17.05.31**
- #1183 hwmon config files for occ, ambient, vrm's
  * 6/12: Got all hwmon-related changes built, tested them and hit an issue, need to debug. Today need to do end-to-end testing for all OCC drivers.

- #343 Power cap

- #1161 Witherspoon: System Failed to respond from 2nd iteration and serial console hangs
  * 6/12: Debugging but couldn't recreate.

---

**Michael:mtritz -**

**17.04.05**
  * 6/09: Almost done addressing code review comments.

** 17.05.17**
- openbmc/openbmc-test-automation#358 [CT Metrics] Add more fields to the Metrics database

**17.05.31**
- #1716 PNOR: Wait for Activation service files to finish
  * 6/12: Continuing.
- #1760 generate-squashfs: Default file name improvements
  * 6/12: Pushed code for review

**17.06.14**
- #1554 PNOR: Implement Software ActivationProgress.interface
---

**Saqib**

**17.04.19**
- openbmc/openbmc-test-automation#511 CodeUpdate: Test for REST upload

**17.05.17**
- openbmc/openbmc-test-automation#572 CodeUpdate: Test End to End Code Update Path

  * 6/07: Pushed code for code-review story #1715 

**17.05.31**
- #1553 PNOR: Implement Software RedundancyPriority.interface
  * 6/09: Made more progress.
- #1715 BMC: Implement Common.FilePath interface
  * 6/12: Made changes per code review comments.

**17.06.14**
- #1762 PNOR: On BMC boot, create the Software version that points to the current PNOR image
  * 6/12 Made initial changes.
- #1755 BMC: Implement Software RedundancyPriority.interface
---

**Leo**

**17.03.22**
- #1360 CI dependency for phosphor-logging
  * 6/01: Continuing working on this one.
  * 6/05: Pushed for review.
  * 6/06: Merged!

- #1367 REST login intermittent fails
  * 6/05: Will start.
  * 6/12: Got review comments.

**17.04.05**
- #1160 phosphor-rest-server: Set property with UInt64 type not supported
  * 6/01: Ready for merge review.
  * 6/02: Continuing, need to add dependency tree structure.
  * 6/05: Got comment from Brad to make minor change.
  * 6/05: Merged!

**17.05.31**
- #1718 REST upload failed : IOError: [Errno 28] No space left on device
  * 6/06: Will start debug, high priority issue.
  * 6/08: Figured out the problem: gevent wasn't closing the file descriptors.
  * 6/07: Continuing debug
  * 6/12: Up for review.

closed 2017-07-10 by zahrens