openbmc/openbmc #1750: Clear SMS attention during power on

geissonator, 2017-06-07:
The SMS attention is now used on all power off's to give the host a chance to power down (soft power off).  There are times where the host is not in a state to respond to that attention though.  The attention will stay asserted though through the next boot, at which point the host will see it and respond, and the BMC will have no idea what it's looking for.  BMC should just be sure to clear the attention during the chassis poweron target.

geissonator, 2017-06-07:
On further analysis of the code, I have no idea if this is possible from the BMC with the current driver, pinged Alistair for some advice.

geissonator, 2017-06-08:
Alistair says it doesn't look promising for a mechanism to clear the attention on the BMC side.  Going to have to pivot this story to just handling situations where the host responds to an SMS and we don't have anything in the queue.  Basically, just return nothing and the host will be fine.

rfrandse, 2017-06-08:
<https://gerrit.openbmc-project.xyz/4651> Return the hearbeat command if queue is empty
Resolves  #1750 Clear SMS attention during power on

closed 2017-06-12 by williamspatrick