openbmc/openbmc #1752: Reduce mboxd logging

geissonator, 2017-06-08:
This was probably needed when first bringing up the mboxd function but now that's it's matured, I think it's time to remove some of this verbose logging, tends to fill the journal up.

```
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.871579765] Host requested flash @ 0x00510000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.872087118] Window @ 0x73738000 for size 0x00100000 maps flash offset 0x00500000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.872646351] Writing MBOX response: 1
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.892142997] Received MBOX command: 4
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.892729704] Host requested flash @ 0x00680000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.893190310] Window @ 0x73438000 for size 0x00100000 maps flash offset 0x00600000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.893686431] Writing MBOX response: 1
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.896335846] Received MBOX command: 4
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.896941906] Host requested flash @ 0x00510000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.897454835] Window @ 0x73738000 for size 0x00100000 maps flash offset 0x00500000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.898068613] Writing MBOX response: 1
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.901523724] Received MBOX command: 4
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.902256128] Host requested flash @ 0x00120000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.902731199] Window @ 0x73838000 for size 0x00100000 maps flash offset 0x00100000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.903228371] Writing MBOX response: 1
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.912499805] Received MBOX command: 4
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.913328613] Host requested flash @ 0x00500000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.913852128] Window @ 0x73738000 for size 0x00100000 maps flash offset 0x00500000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844958.914533300] Writing MBOX response: 1
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.059604007] Received MBOX command: 4
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.060331118] Host requested flash @ 0x00890000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.061049300] Window @ 0x73638000 for size 0x00100000 maps flash offset 0x00800000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.061599603] Writing MBOX response: 1
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.066956048] Received MBOX command: 4
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.067518633] Host requested flash @ 0x007a0000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.068037623] Window @ 0x73538000 for size 0x00100000 maps flash offset 0x00700000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.068599805] Writing MBOX response: 1
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.079744977] Received MBOX command: 4
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.080472169] Host requested flash @ 0x00520000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.080934512] Window @ 0x73738000 for size 0x00100000 maps flash offset 0x00500000
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.081560654] Writing MBOX response: 1
Jun 07 14:16:04 zaius mboxd[846]: [ 1496844959.084452330] Received MBOX command: 4
Jun 07 14:16:05 zaius obmc-console-server[845]: obmc-console-server: Write error: Resource temporarily unavailable
Jun 07 14:16:05 zaius mboxd[846]: [ 1496844959.085665906] Host requested flash @ 0x01580000
Jun 07 14:16:05 zaius mboxd[846]: [ 1496844959.462894431] Window @ 0x74038000 for size 0x00100000 maps flash offset 0x01500000
Jun 07 14:16:05 zaius mboxd[846]: [ 1496844959.463078149] Writing MBOX response: 1
Jun 07 14:16:05 zaius mboxd[846]: [ 1496844959.477240411] Received MBOX command: 4
Jun 07 14:16:05 zaius mboxd[846]: [ 1496844959.477407846] Host requested flash @ 0x00890000
Jun 07 14:16:05 zaius mboxd[846]: [ 1496844959.477516977] Window @ 0x73638000 for size 0x00100000 maps flash offset 0x00800000
Jun 07 14:16:05 zaius mboxd[846]: [ 1496844959.477614674] Writing MBOX response: 1
...
```

amboar, 2017-06-09:
What happened was the meaning of `--verbose` was changed, which increased the verbosity for no change in the unit file.

amboar, 2017-06-09:
https://gerrit.openbmc-project.xyz/#/c/4696/

rfrandse, 2017-06-09:
<https://gerrit.openbmc-project.xyz/4696> mboxd: Reduce logging verbosity
Fixes  #1752 Reduce mboxd logging

closed 2017-06-12 by williamspatrick