openbmc/openbmc #1759: BMC shows constant loadavg of 1.0 even while idle

rlippert, 2017-06-08:
BMC shows a constant loadavg of >1 even when not doing anything:

```
# uptime
 20:55:44 up  2:21,  load average: 1.00, 1.00, 1.00
```

I think the issue is that the fsi_master_ipol thread is sleeping in the wrong state and still counted as runnable:

```
# cat /proc/1255/status
Name:   fsi_master_ipol
Umask:  0000
State:  D (disk sleep)
Tgid:   1255
```

FYI @shenki

shenki, 2017-06-09:
@rlippert What's your system version? and kernel version?

I assume you're on v4.7. The fsi_master_ipol code is not present in the 4.10 branch, so you should not see it once you're moved to there.

Should @cbostic and his team propose to merge it upstream, or in 4.10, I will ensure it goes through a through code review cycle to catch issues like this.

rlippert, 2017-06-12:
Yes we are still using v4.7 kernel.  Will retest once we have a working v4.10 kernel.

cbostic, 2017-10-16:
Are you still seeing the same values on dev-4.10 kernel?   Will check some test systems today.

Marking as 'Phase 7'

cbostic, 2017-10-20:
load average appears to be lower now.  

```
root@zaius:~# uptime
 15:07:51 up 45 min,  load average: 0.06, 0.16, 0.35
```

cbostic, 2017-10-20:
@rlippert given the load average is better now so will close.  Please reopen if you feel this is still a problem.  @rfrandse can you close?

rlippert, 2017-10-20:
We still have not been able to move to 4.10 based BMC kernel yet so I haven't had a chance to verify.

You can close if it looks good on your side.

gtmills, 2018-08-20:
Closing. Please reopen if still a problem. thanks!

closed 2018-08-20 by gtmills